google/aiyprojects-raspbian #537: get the input when run CameraInference

why702, 2018-11-29:
If I do the camera.capture after applying the CameraInference, will get the captured image with some frame delay.
Is it possible to get the input frame when I get the result of inference.run?
ex:
`for result, **input** in inference.run(num_frames): ...`

zye1996, 2018-12-04:
Try put the capture into a thread. Refer to the face_detection example

dmitriykovalev, 2018-12-06:
@why702 That's problematic for real-time performance. Raspberry Pi receives images via MIPI interface (fast) and inference data via SPI interface (slow). Sending image data over SPI is possible but would require a lot of time (we used that for debug internally). What's your use case?

eld1e6o, 2019-01-28:
Hello! I have the same issue.
As example, running the face detector example, if the face is moving fastly, the bounding box drawn is misplaced (it seems to be from an older frame).
Then, if I try to crop the bounding box of the detected object, it must be moving slowly.
There is some solution?