google/aiyprojects-raspbian #542: Voice Kit - assistant demo not working?

rutabaga76, 2018-12-04:
I'm a total newbie here. I'm a teacher, and have been trying to work through setting up the AIY Voice Kit with a few of my Intro to CS students this week. We made it through all the set up instructions online, the speaker and microphone tests worked fine. When we get to the step to "run the assistant demo", however, it did not run. It would error messages about lines of code. I didn't screenshot it at the time, but it wasn't too dissimilar from the error messages we are getting now (see below).
We had to leave it for the day, and this morning I loaded it up and the kit wouldn't connect - so I reflashed the SD card by downloading the image from the aiyprojects website, and I used Etcher.io to write the new image on the SD card.

I went through the logging in steps again - I was able to connect, I ran the tests for microphone and speakers successfully. 

Back to the step to run the assistant demo. 
I typed the command line: src/examples/voice/assistant_grpc_demo.py

It runs for a while, then returns the following message:

pi@raspberrypi:~/AIY-voice-kit-python $ src/examples/voice/assistant_grpc_demo.py
Traceback (most recent call last):
  File "src/examples/voice/assistant_grpc_demo.py", line 57, in <module>
    main()
  File "src/examples/voice/assistant_grpc_demo.py", line 49, in main
    language_code=args.language)
  File "/opt/aiy/projects-python/src/aiy/assistant/grpc.py", line 205, in __init__
    super().__init__(language_code, volume_percentage)
  File "/opt/aiy/projects-python/src/aiy/assistant/grpc.py", line 64, in __init__
    credentials = auth_helpers.get_assistant_credentials()
  File "/opt/aiy/projects-python/src/aiy/assistant/auth_helpers.py", line 132, in get_assistant_credentials
    return _try_to_get_credentials(credentials_file)
  File "/opt/aiy/projects-python/src/aiy/assistant/auth_helpers.py", line 123, in _try_to_get_credentials
    credentials = _credentials_flow_interactive(client_secrets)
  File "/opt/aiy/projects-python/src/aiy/assistant/auth_helpers.py", line 70, in _credentials_flow_interactive
    scopes=[_ASSISTANT_OAUTH_SCOPE])
  File "/usr/local/lib/python3.5/dist-packages/google_auth_oauthlib/flow.py", line 172, in from_client_secrets_file
    client_config = json.load(json_file)
  File "/usr/lib/python3.5/json/__init__.py", line 268, in load
    parse_constant=parse_constant, object_pairs_hook=object_pairs_hook, **kw)
  File "/usr/lib/python3.5/json/__init__.py", line 319, in loads
    return _default_decoder.decode(s)
  File "/usr/lib/python3.5/json/decoder.py", line 339, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "/usr/lib/python3.5/json/decoder.py", line 355, in raw_decode
    obj, end = self.scan_once(s, idx)
json.decoder.JSONDecodeError: Expecting property name enclosed in double quotes: line 2 column 1 (char 2)

What can we do to make this work? My students, and myself, are feeling frustrated - but I'm using this as  a lesson in problem solving and persistance! Any and all help is greatly appreciated! (remembering I'm a newbie to raspberry pi). Thank you in advance!

rutabaga76, 2018-12-04:
Okay, I solved my own problem. I'm working on a Chromebook and had to use JSON viewer to open my credentials file. I was looking through the tutorial and noticed that my credentials file text did not look like that of the sample one given - mine was missing double quotes around "installed", "client_id", etc... so I went in and added quotes (the last line of the error message about expecting double quotes tipped me off...)
So, that works now! We are struggling getting the button to work... no response. But that is another problem perhaps for another thread. Going to double check our connections first.

closed 2018-12-04 by rutabaga76

epaph, 2019-11-23:
i am having a similar issue, but i checked the quotes and didnt seem to have the same problem there- any ideas why the assistant.json file might not be working?