google/aiyprojects-raspbian #543: ImportError: No module named 'aiy.pins'

darrenchiu, 2018-12-07:
I am trying to control the LED with a very simple script but it doesn't seem to be able to import from `aiy.pins`. Could anyone shed some light on this?

my simple script:
```
import aiy.audio
import aiy.cloudspeech
import aiy.voicehat

from time import sleep
from gpiozero import LED
from aiy.pins import LED_1

led = LED(LED_1)
while True:
                 led.on()
                sleep(1)
                led.off()
                sleep(1)

```

Onewayfong, 2018-12-26:
Darren, do you solve the issue now? because we have a similar problem. it warns: no aiy.vision.pin. would you like to give some instruction? thanks.

dmitriykovalev, 2018-12-27:
@Onewayfong Do you use the latest SD card image from  https://github.com/google/aiyprojects-raspbian/releases ?

Onewayfong, 2018-12-28:
@dmitriykovalev Yes. we use the latest SD card image. and we also change the pin name from aiy.vision to aiy. it didn't work. and we found there is something wrong with the PI. after we urgrade it, it is okay now. thanks for your attention.

dmitriykovalev, 2018-12-28:
@Onewayfong Sounds good!

closed 2018-12-28 by dmitriykovalev