google/aiyprojects-raspbian #544: import error :no module name aiy.audio

samysinhg, 2018-12-10:
i am trying to run servo motors with the help of instructions given in magiPi but when i run the python it shows the  error

import error :no module name aiy.audio

samysinhg, 2018-12-10:
/home/samy/Pictures/Screenshot from 2018-12-10 03-18-57.png

samysinhg, 2018-12-10:
[image]

LizMyers, 2018-12-12:
+1 same issue. The problem started when I attempted to uninstall/reinstall a fresh disk image. Magpi link downloads the latest from GitHub - which doesn't seem compatible with earlier demo files that import the aiy.audio module and aiy.speechhat

samysinhg, 2018-12-13:
@LizMyers you are right !
my problem of aiy.audio is solved when i import the new updated python library as aiy.voice.audio
try this solution and let me know and for many other libraries name is changed!

LizMyers, 2018-12-13:
@samysinhg Hey! So Import aiy.voice.audio works! Now I have another error: 
AttributeError: module 'aiy.cloudspeech' has not attribute 'get_recognizer'

Do you know what 'get_recognizer' has been replaced with? I'm headed over to the API docs now...

LizMyers, 2018-12-13:
On another note - am now exploring demos in the new image - and the **Assistant Sound Quality has improved a ton.**  It seems like the API has changed significantly too - guess I need to update myself and my scripts to get stuff working again. Do you know if there's a bridge tutorial - "... ok, so you were using 'get_recognizer()' before and now we use XYZ ... " ??

PichuChen, 2018-12-21:
Refer to https://github.com/google/aiyprojects-raspbian/blob/aiyprojects/src/examples/voice/cloudspeech_demo.py

I will use `client = CloudSpeechClient()`

LizMyers, 2018-12-24:
Anyone know how to play a wav file? According to the API aiy.voice.audio.play_wav('myfilename') should work - but doesn't.

dmitriykovalev, 2018-12-27:
@LizMyers Please take a look at our new API to play `.wav` files: https://aiyprojects.readthedocs.io/en/latest/aiy.voice.audio.html