google/aiyprojects-raspbian #545: Trigger my Google action on pressing a button

fabbbbb, 2018-12-10:
I have a AIY Voice Kit fully assembled and functional (I have tested the 3 demos.py sucessfully). 
I managed to launch the demo where we have to push the button to launch the Google assistant. 
But now I wan that it ONLY trigger my Google action (and directly launch « Welcome to XXXX app. Do you want… »). 
How can I manage to do that ?

dmitriykovalev, 2018-12-11:
What do you mean by "my Google action"? Please provide more details.

fabbbbb, 2018-12-12:
> What do you mean by "my Google action"? Please provide more details.

[https://developers.google.com/actions/](url)

dmitriykovalev, 2018-12-12:
https://developers.google.com/assistant/sdk/guides/library/python/extend/custom-actions
https://developers.google.com/assistant/sdk/guides/service/python/extend/custom-actions

fabbbbb, 2019-01-14:
Do I need to create a custom action or can I only modify textinput.py our pushtotalk.py ? 
Which python code I have to insert?

fabbbbb, 2019-01-14:
> https://developers.google.com/assistant/sdk/guides/library/python/extend/custom-actions
> https://developers.google.com/assistant/sdk/guides/service/python/extend/custom-actions

Do I need to create a custom action or can I only modify textinput.py our pushtotalk.py ?
Which python code I have to insert?

fabbbbb, 2019-01-14:
Maybe using https://github.com/google/aiyprojects-raspbian/blob/aiyprojects/src/examples/voice/assistant_library_with_local_commands_demo.py ?

dmitriykovalev, 2019-01-16:
@fabbbbb I think you should just follow the steps described on web pages and modify our example code. Unfortunately that's not a one line code change.

fabbbbb, 2019-01-17:
@dmitriykovalev 
I manage to do that with one line code : `self._assistant.send_text_query("what time is it")`

dmitriykovalev, 2019-01-18:
I thought we were discussion device actions but if `self._assistant.send_text_query("what time is it")` is exactly what you need then this is solved!

closed 2019-01-18 by dmitriykovalev