google/aiyprojects-raspbian #548: strange error

jsophil, 2018-12-20:
when i test a demo.py( actually the official support: assistant_library_demo.py )
the python program take me to a website like below:
[image]

when i enter in
[image]

when i click the allow button，i meet an error 
[image]

please help me ( i am really anxious )

manoj7410, 2018-12-20:
Did you also try assistant_grpc_demo.py ?

jsophil, 2018-12-20:
> Did you also try assistant_grpc_demo.py ?

no， i haven't

jsophil, 2018-12-20:
how could i do,i am really anxious

mongonta0716, 2019-01-19:
Hi, @jsophil 
This phenomenon occurs when copying the displayed URL after running assistant_library_demo.py via SSH and opening the URL in a browser on another PC. You can authenticate by connecting to RaspberryPi with VNC or by using SSH with X11 Forwarding.