google/aiyprojects-raspbian #553: Accessing the SAM D09's 12-bit ADC

Exscotticus, 2018-12-26:
The Vision Bonnet exposes some of the SAM D09's analog pins (PA04 and PA05). Is there any Python API examples of reading voltage values from these pins?

dmitriykovalev, 2018-12-27:
You can use kernel [IIO](https://www.kernel.org/doc/html/v4.14/driver-api/iio/index.html) subsystem to read voltages:
```bash
cat /sys/bus/iio/devices/iio\:device0/in_voltage0_user0_input
cat /sys/bus/iio/devices/iio\:device0/in_voltage1_user1_input
cat /sys/bus/iio/devices/iio\:device0/in_voltage2_user2_input
cat /sys/bus/iio/devices/iio\:device0/in_voltage3_user3_input
```

Exscotticus, 2018-12-27:
I just confirmed this works for PA04 (A). Cool! And thank you...

I'm surprised you can use PA10 (C) and PA11 (D) for analog input. According to the SAM D09 datasheet, those are digital pins.

dmitriykovalev, 2018-12-27:
Datasheet actually looks correct:

[image]

Exscotticus, 2018-12-27:
You're right. The datasheet I was looking at had a diagram that led me to believe those were digital-only pins.

[image]

dmitriykovalev, 2018-12-28:
Alright, I guess this can be closed :)

closed 2018-12-28 by dmitriykovalev