google/aiyprojects-raspbian #554: Button / LED / Form factor

HUGEwindow, 2018-12-29:
Is it possible to use the pi and vision bonnet without button, led, etc ?  Are these required? 

Thanks in advance.

divx118, 2019-01-07:
@HUGEwindow I don't know the vision bonnet, but for the voicekit I build my own. That can be done without button and leds. I used my own usb soundcard and speakers with amplifier to have stereo sound. 
I think the same can be done for the vision, but I think you will need the camera, however led and button should not be needed. The led only shows the status of the assistant, talking listening ready etc ...
The button is normally the equivalent for saying ok google. Depends on which example you run of course, if you can leave it out.

dmitriykovalev, 2019-01-08:
@HUGEwindow For sure you can do that. You can even use it without a camera and run inference only on image files. I guess that depends on your project :)