google/aiyprojects-raspbian #556: Running auto update after flashing 11.16 breaks camera sensor

branflake2267, 2019-01-01:
Camera sensor fails to work after updating the system. Seems the drivers that are updated break operations. 

Reproduction:
1. Flash 11.16
2. Boot up system
3. Reset password to pi, add wifi, set location...
4. Wait for auto-update prompt and say ok
5. Wait for auto-update to finish hour(s) later...
6. Try joy demo, or  camera detection demo, or raspistill -o test.png - it won't work

branflake2267, 2019-01-01:
Looks like this duplicates: https://github.com/google/aiyprojects-raspbian/issues/527

branflake2267, 2019-01-01:
By the way, I did take it apart and check all the cable connections and that wasn't it. The reason I could tell is I re-flashed the sd card and booted again.

branflake2267, 2019-01-02:
I did another update. I've found that `raspistill -o test.jpg` is working but the `face_detection_camera.py` won't work, and it complains about the /dev/vision_spicomm is missing.

branflake2267, 2019-01-02:
Related: https://github.com/google/aiyprojects-raspbian/issues/382

branflake2267, 2019-01-02:
Related: https://github.com/google/aiyprojects-raspbian/issues/377 

I get the same error in the kern.log `aiy-vision spi0.0: Incoming crc mismatch: slave ec6zef0 vs computed e2b16736`...

nicholst, 2019-02-26:
Same happens to me, but it kills the even `raspistill` generating either this error
```Camera control callback  cmd=0x4f525245mmal: No data received from sensor. Check all connections, including the Sunny one on the camera board```
or a repetition of
```Message from syslogd@raspberrypi at Feb 26 17:15:48 ...
 kernel:[  312.650945] Code: bad PC value```

manoj7410, 2019-11-14:
This issue seems to be fixed with the latest AIY image : https://github.com/google/aiyprojects-raspbian/releases/tag/v20191113.

closed 2020-11-20 by dmitriykovalev