google/aiyprojects-raspbian #557: buzzer_tracker_demo.py will only plays the congratulations.track correctly

branflake2267, 2019-01-01:
Running the buzzer_tracker_demo.py only runs the congratulations.track correctly. The other tracks will only play one beep. Version 11.16 and without the latest system update.

Reproduction
1. cd ./examples/buzzer/
2. run `buzzer_tracker_demo.py dramatic.track`
3. It plays one beep.

albellows, 2019-04-17:
This is also true on my Vision kit (not just a Voice issue).  The other demo, buzzer_demo.py, works just fine though.  So I'm guessing it's either a bug in the buzzer_tracker_demo code or in the aiy.trackplayer package (TrackPlayer or TrackLoader).

Franchie, 2020-03-14:
Probably resolved by now, but this is caused by a remnant of what appears to be a debug statement, which prevents processing tracks with commands (exits when one of these is encountered). If you remove the lines 343..345 in trackplayer.py, this then works normally:
https://github.com/google/aiyprojects-raspbian/blob/aiyprojects/src/aiy/trackplayer.py#L343...L345

aklingam7, 2020-08-20:
I've noticed the same issue with my Vision kit. As soon as I got rid of the lines in tackplayer.py it started working.