google/aiyprojects-raspbian #562: Make TTS Match Assistant Quality/Tone

LizMyers, 2019-01-07:
The current TTS voice is very muffled and sometimes difficult to understand. For further context, please see this [30 sec. demo](https://www.youtube.com/watch?v=NAiJghr5l6E). 

My problem is that I want to be able to use the API with the **same audio quality as with the Assistant** or other GRPC demoes included with the AIY-Projects image.  

BTW - I find it confusing from a UX standpoint that there are demos with different agents and mixed audio quality included with the voice kit. Sometimes I hear a US male voice, sometimes there's a UK female UK voice, and when using tts - the muffled robotic voice.

Thanks in advance for your time and attention.

dmitriykovalev, 2019-01-08:
We currently use [pico](https://en.wikipedia.org/wiki/SVOX) tts engine. It works on-device  and doesn't require internet connection. There are other on-device tts python libraries available.

As an option you can try to use [Google Cloud Text-to-Speech API](https://cloud.google.com/text-to-speech/docs/basics) which requires internet connection.  We are going to provide Python code example with the next SD card image release.

LizMyers, 2019-01-08:
Thanks Dmitriy - I'm looking forward. My project works with Google Cloud so requires internet anyway.  Does TTS suffer latency problems? Why the emphasis of on device tts?

dmitriykovalev, 2019-01-09:
@LizMyers There are no latency issues, it's up  to you which tts engine to use. If you are already using internet connection then it's perfectly fine!

olafthiele, 2019-06-09:
Is there any Python code already available to use the cloud tts? Couldn't find any