google/aiyprojects-raspbian #566: Question: using/converting models for use with TensorFlow

edrdo, 2019-01-15:
I have a few questions regarding the AIY Vision Kit in particular regarding VisionBonnet  (sorry if some most of them go beyond strict development issues, but this seems to be the better place to task):

* Does the VisionBonnet chip use TensorFlow? 
* Either way, is it possible to convert the models available at the [AIY projects]( https://aiyprojects.withgoogle.com/models/) for use with TensorFlow in other settings (e.g. using the TF API directly) ?
* Is  the standalone VisionBonnet chip available for sale and how does it releate to the upcoming [Edge TPU Devices](https://aiyprojects.withgoogle.com/edge-tpu/) ?

edrdo, 2019-01-16:
Thanks for the answers.

>  Some compute graphs can be converted from TensorFlow by using our compiler.

I see, but I was looking for a way to accomplish the reverse operation, i.e., to convert a Vision Kit model to TensorFlow format. In particular I was interested in doing so for the [Nature Explorer model](https://aiyprojects.withgoogle.com/model/nature-explorer/).

weiranzhao, 2019-01-16:
@edrdo 

Unfortunately, there's no method to do the reverse, during the conversion, some information is lost.

closed 2019-02-02 by edrdo