google/aiyprojects-raspbian #568: AIY v1 support Pi 3 A+ model?

avishek-rc1, 2019-01-17:
Please help. Does the AIY v1 support Pi 3 A+ model? I have been trying with the latest image and still no luck.

dmitriykovalev, 2019-01-18:
@avishek-rc1 Do you mean VoiceHAT on Pi 3A+ ? It should work. What kind of problem you have? Please provide more details.

andrewtvuong, 2019-01-21:
@avishek-rc1 @dmitriykovalev Sorry not sure if I should open a new issue. for the Pi 3A+ it seems assistant_library_with_button_demo.py and assistant_library_with_local_commands_demo.py hangs, while grpc commands still work. I've reimaged twice. 

When it doesn't hang none of the services work including the grpc commands, I get nothing. It doesn't register what I say to it. I get the following message:

```
DEBUG:google.auth.transport.requests:Making request: POST https://www.googleapis.com/oauth2/v3/token
DEBUG:urllib3.connectionpool:Starting new HTTPS connection (1): www.googleapis.com:443
DEBUG:urllib3.connectionpool:https://www.googleapis.com:443 "POST /oauth2/v3/token HTTP/1.1" 200 None
INFO:aiy.assistant.grpc:Connecting to embeddedassistant.googleapis.com
/opt/aiy/projects-python/src/aiy/board.py:218: RuntimeWarning: This channel is already in use, continuing anyway.  Use GPIO.setwarnings(False) to disable warnings.
  GPIO.setup(channel, GPIO.OUT)
INFO:root:Press button to start conversation...
INFO:root:Conversation started!
INFO:aiy.assistant.grpc:Recording started.
INFO:aiy.assistant.grpc:End of audio request detected.
DEBUG:aiy.assistant.grpc:Updating conversation state.
INFO:aiy.assistant.grpc:Not expecting follow-on query from user.
INFO:aiy.assistant.grpc:Recording stopped.
aplay: playback:2787: read error
INFO:root:Press button to start conversation...
```

siennx, 2019-02-13:
me cannot get the latest aiy image to work with 3A+ neither.  When I put the same sd card to zero or 3B+, it works but not with 3A+.  The image cannot boot with 3A+ for some reason.  I am sure my 3A+ board is working fine with other images.  This image simply does not run with 3A+.  Any one could help?

andrewtvuong, 2019-02-13:
@siennx same cannot get it voice hat v1 to work with raspberry pi 3A+ unfortunately. It's weird because it looked like the voice hat was designed for the 3A+ based on appearance.

andrewtvuong, 2019-02-21:
@siennx @avishek-rc1 I have figured out how to get 3A+ to work with voicehat v1. Install latest Raspbian 11-13-18 on your own from raspberrypi.org. 

Then follow a bit of the instructions from https://github.com/google/aiyprojects-raspbian/blob/aiyprojects/HACKING.md starting with:

`To update your system image, download the latest .img.xz file from our releases page. Once downloaded, write the image to your SD card, and you're good to go!`

**DO NOT FLASH WITH IMAGE**
simply go to the [release page](https://github.com/google/aiyprojects-raspbian/releases) to 4-13-18 and download the zip or tar.

Follow these steps 

AIY Debian Package Repo
Add AIY package repo:

`echo "deb https://dl.google.com/aiyprojects/deb stable main" | sudo tee /etc/apt/sources.list.d/aiyprojects.list`
Add Google package keys from https://www.google.com/linuxrepositories/:

`wget -q -O - https://dl.google.com/linux/linux_signing_key.pub | sudo apt-key add -`
if the above doesn't work. download the key.pub file separately then `sudo apt-key add linux_signing_key.pub`

Update and install the latest system updates (including kernel):

`sudo apt-get update
``sudo apt-get upgrade
`Reboot after update:

`sudo reboot
`
unzip or tar the aiyprojects-raspbian-20180413.zip,tar then move into it
`cd aiyprojects-raspbian-20180413/scripts/`
then run the scripts in the following order:
`install-alsa-config.sh`
`install-services.sh`

change the boot file and comment out everything existing
`sudo nano /boot/config.txt`
add the following
```
dtoverlay=dwc2
start_x=1
gpu_mem=128
dtoverlay=googlevoicehat-soundcard
```
reboot and audio should work on 3A+ and v1 voicehat. cheers let me know if you have any issues.