google/aiyprojects-raspbian #570: Files to load kennel modules see systemctl status systemd-modules-load.service provides no clues except to show modules are not loaded!

rusty377, 2019-01-20:


dmitriykovalev, 2019-01-25:
It's not clear from your description what kind of problem you have. Please provide more information, e.g. kernel module names.

tarzanek, 2019-03-14:
quick fix is:
`apt-get install --reinstall pwm-soft-dkms
systemctl status systemd-modules-load.service

systemctl start systemd-modules-load.service
`

tarzanek, 2019-03-14:
that said ... @dmitriykovalev I think with new update of raspbian the kernel is now v7+
"Building for 4.14.98+ and 4.14.98-v7+ "

so it seems all aiy dkms drivers don't get rebuilt upon latest dist-upgrade ...
so all these would need a rebuild/reinstall:

root@raspberrypi:~# apt-cache search aiy |grep dkms
aiy-dkms - Hardware device drivers for AIY Projects Kits.
aiy-vision-dkms - AIY Vision DKMS driver.
aiy-voicebonnet-soundcard-dkms - AIY Voice DKMS driver.
leds-ktd202x-dkms - Hardware device drivers for AIY Projects Kits.
pwm-soft-dkms - Hardware device drivers for AIY Projects Kits.

@dmitriykovalev could it be the dkms modules won't get picked up?