google/aiyprojects-raspbian #576: "git checkout voicehat" not working

MaiMariarti, 2019-02-08:
I issued "git checkout voicehat" yesterday and it worked.
Today, I get the message "error: pathspec 'voicehat' did not match any file(s) known to git."
I am using AIY kit version 1 if that's of any importance. 
It is supposed to create two scripts in 
scripts/configure-driver.sh
scripts/install-alsa-config.sh
as per this page https://medium.com/snips-ai/is-it-google-aiy-nah-its-snips-f67c9dc2139a
under section (1- Configure Google AIY Voice Hat)
Any ideas why this is happening?

dmitriykovalev, 2019-02-23:
Looks like you are using pretty old SD card image, please download the latest one at https://github.com/google/aiyprojects-raspbian/releases

closed 2019-11-14 by dmitriykovalev

MaiMariarti, 2019-11-14:
I threw the whole thing in garbage where it belongs.