google/aiyprojects-raspbian #577: Internal Server Error

CarlP66, 2019-02-09:
AIY Voice has been working fine for months. No changes on my side, but today when I run the python script that I have been using, I get:

"requests.exceptions.HTTPError 500 Server Error: Internal Server Error for url: https://embeddedassistant.googleapis.com/v1alpha2/projects/voice-assistant-213123/deviceModels/voice-assistant-213123-voice-kit"

This looks very much like an issue at Google's end. Can someone confirm, and advise when it will be back up.

Thanks, Carl.

CarlP66, 2019-02-09:
Seems OK now. I'm in no rush to close it though - lets see if it stays fixed.

dmitriykovalev, 2019-02-23:
Similar problems already happened before for other users. Looks like a server-side issue. Please reopen if that still happens.

closed 2019-02-23 by dmitriykovalev

CarlP66, 2019-05-12:
Need to re-open this - I am having a repeat occurrence.  As per comment from dmitriykovalev.