google/aiyprojects-raspbian #578: Syntax error with Assistant_grpc_demo.py line 6

ajmcqueen, 2019-02-13:
I'm trying to setup my project for the first time and i'm seeing this issue when I run the assistant_grpc_demo.py script:

```
Traceback (most recent call last):
  File "/usr/local/bin/assistant_grpc_demo.py", line 6, in <module>
    exec(compile(open(__file__).read(), __file__, 'exec'))
  File "/opt/aiy/projects-python/src/examples/voice/assistant_grpc_demo.py", line 20, in <module>
    import aiy.assistant.grpc
  File "/opt/aiy/projects-python/src/aiy/assistant/grpc.py", line 17, in <module>
    import aiy._apis._speech
  File "/opt/aiy/projects-python/src/aiy/_apis/_speech.py", line 30, in <module>
    from google.cloud import speech
  File "/usr/local/lib/python3.5/dist-packages/google/cloud/speech/__init__.py", line 22, in <module>
    from google.cloud.speech.client import Client
  File "/usr/local/lib/python3.5/dist-packages/google/cloud/speech/client.py", line 25, in <module>
    from google.cloud.speech._gax import GAPICSpeechAPI
  File "/usr/local/lib/python3.5/dist-packages/google/cloud/speech/_gax.py", line 17, in <module>
    from google.cloud.gapic.speech.v1.speech_client import SpeechClient
  File "/usr/local/lib/python3.5/dist-packages/google/cloud/gapic/speech/v1/speech_client.py", line 31, in <module>
    from google.gapic.longrunning import operations_client
  File "/usr/local/lib/python3.5/dist-packages/google/gapic/longrunning/operations_client.py", line 50, in <module>
    from google.longrunning import operations_pb2
  File "/usr/local/lib/python3.5/dist-packages/google/longrunning/operations_pb2.py", line 23, in <module>
    from google.longrunning.operations_grpc_pb2 import *
  File "/usr/local/lib/python3.5/dist-packages/google/longrunning/operations_grpc_pb2.py", line 341, in <module>
    from grpc.beta import implementations as beta_implementations
  File "/usr/local/lib/python3.5/dist-packages/grpc/beta/implementations.py", line 22, in <module>
    from grpc import _auth
  File "/usr/local/lib/python3.5/dist-packages/grpc/_auth.py", line 17, in <module>
    from concurrent import futures
  File "/usr/local/lib/python3.5/dist-packages/concurrent/futures/__init__.py", line 8, in <module>
    from concurrent.futures._base import (FIRST_COMPLETED,
  File "/usr/local/lib/python3.5/dist-packages/concurrent/futures/_base.py", line 381
    raise exception_type, self._exception, self._traceback
                        ^
SyntaxError: invalid syntax
```

I'm completely new to this and I'm not sure what the issue might be.

dmitriykovalev, 2019-02-23:
You are probably using python2, you need to use python3.

manoj7410, 2019-02-25:
@ajmcqueen Did you run the command as "assistant_grpc_demo.py" or "./assistant_grpc_demo.py" after entering into : /home/pi/AIY-voice-kit-python/src/examples/voice ?

In your logs, it says you are executing an assistant demo from PATH : "File "/usr/local/bin/assistant_grpc_demo.py", line 6, in <module>"

You can run below command : 

`/home/pi/AIY-voice-kit-python/src/examples/voice/assistant_grpc_demo.py` to get rid of that error and proceed ahead.

downeyjar, 2020-02-13:
/home/pi/AIY-voice-kit-python/src/examples/voice/assistant_grpc_demo.py
Traceback (most recent call last):
  File "/home/pi/AIY-voice-kit-python/src/examples/voice/assistant_grpc_demo.py", line 57, in <module>
    main()
  File "/home/pi/AIY-voice-kit-python/src/examples/voice/assistant_grpc_demo.py", line 49, in main
    language_code=args.language)
  File "/home/pi/AIY-projects-python/src/aiy/assistant/grpc.py", line 258, in __init__
    super().__init__(language_code, volume_percentage)
  File "/home/pi/AIY-projects-python/src/aiy/assistant/grpc.py", line 94, in __init__
    credentials = auth_helpers.get_assistant_credentials()
  File "/home/pi/AIY-projects-python/src/aiy/assistant/auth_helpers.py", line 151, in get_assistant_credentials
    return _try_to_get_credentials(credentials_file)
  File "/home/pi/AIY-projects-python/src/aiy/assistant/auth_helpers.py", line 123, in _try_to_get_credentials
    credentials = _credentials_flow_interactive(client_secrets)
  File "/home/pi/AIY-projects-python/src/aiy/assistant/auth_helpers.py", line 75, in _credentials_flow_interactive
    webbrowser.register('chromium-browser', None, webbrowser.Chrome('chromium-browser'), -1)
TypeError: register() takes from 2 to 3 positional arguments but 4 were given