google/aiyprojects-raspbian #579: Changing Color of LED button in Voice V2

rockman55, 2019-02-15:
How can I change the color of the Arcade Button Led? I have run the led.py example and I know that there are many options to choose from, currently the color of my button is red. I would like to be able to change the color to something other than red. 
Thanks

dmitriykovalev, 2019-02-23:
Please check out this example code file: https://github.com/google/aiyprojects-raspbian/blob/aiyprojects/src/examples/leds_example.py

rockman55, 2019-02-24:
Thanks dmitriykovalev I have run this program, but what I am trying to do is change the color of the button led when I activate the Google Assistant. At this time the button is only red, and I would like to know how to change the color to another color and I am not sure in which program the color is set.
Thanks,
Danny

manoj7410, 2019-02-25:
@rockman55 Can you also have a look at :  https://github.com/google/aiyprojects-raspbian/blob/aiyprojects/src/aiy/board.py
Seems like assistant_library_demo is making use of Board class.

cwadven, 2020-05-23:
If anyone sees, You have to Change https://github.com/google/aiyprojects-raspbian/blob/aiyprojects/src/aiy/board.py "201 line" color changes seems to be (r, g, b)... I took hour to find this....