google/aiyprojects-raspbian #580: camera.capture() not working if Annotator is on

haku15, 2019-02-15:
If I try to capture an image or video from the camera using camera.capture() or camera.start_recording() while the Annotator is on, the application crashes.
This is the code:

```
from aiy.vision.annotator import Annotator
from picamera import PiCamera
import time

def main():

    with PiCamera(sensor_mode=4, framerate=30) as camera:
        
         camera.start_preview()
         annotator = Annotator(camera, dimensions=(320, 240))
         scale_x = 320 / 1640
         scale_y = 240 / 1232

         annotator.clear() 
         annotator.text((10,20),"test")
         annotator.update()

         time.sleep(2)

         camera.capture('test_annotator.jpg')
         '''
         camera.start_recording('my_video.h264')
         camera.wait_recording(3)
         camera.stop_recording()
         '''
         time.sleep(10)

    camera.stop_preview()

if __name__ == '__main__':
    main()
```

The application crashes when camera.capture() gets executed. The error is below and is the same when trying to record a video instead of pic:

```
mmal: mmal_port_pool_destroy: port 0x24a7240, pool 0x24b6970 destroyed while port enabled
Traceback (most recent call last):
  File "ann1.py", line 54, in <module>
    main()
  File "ann1.py", line 43, in main
    camera.capture('test_annotator.jpg')
  File "/usr/lib/python3/dist-packages/picamera/camera.py", line 1418, in capture
    encoder.start(output)
  File "/usr/lib/python3/dist-packages/picamera/encoders.py", line 1126, in start
    super(PiCookedOneImageEncoder, self).start(output)
  File "/usr/lib/python3/dist-packages/picamera/encoders.py", line 377, in start
    self.output_port.enable(self._callback)
  File "/usr/lib/python3/dist-packages/picamera/mmalobj.py", line 1254, in enable
    self._pool.send_all_buffers(block=False)
  File "/usr/lib/python3/dist-packages/picamera/mmalobj.py", line 1941, in send_all_buffers
    super(MMALPortPool, self).send_all_buffers(port, block, timeout)
  File "/usr/lib/python3/dist-packages/picamera/mmalobj.py", line 1892, in send_all_buffers
    self.send_buffer(port, block, timeout)
TypeError: silent_send_buffer() takes 1 positional argument but 4 were given
```
I tried to use annotator.stop() before the camera.capture(), but no change.
Is this due to this [bug](https://github.com/google/aiyprojects-raspbian/blob/56d40c7c1c5125fcfe2e69ec2b6f27a15fe35ad9/src/aiy/vision/annotator.py#L76)?
Any idea what I could do?
I am using a RP3, AIY v20181116, Raspi camera v2(from the Google vision kit), Raspian Stretch 4.14.79-v7+, Python 3.5.3.
Thanks

alew3, 2019-02-22:
I'm having the same problem. Any solutions to this?

samsterckval, 2019-04-03:
Same problem aswell, anyone has a solution yet?

samsterckval, 2019-04-03:
> Same problem aswell, anyone has a solution yet?

It does seem to be fixed when I change the monkey patch to MMALPool.send_buffer and hardcode the args instead of the MMALPortPool.send_buffer.  ( it is the MMALPool.send_buffer that throws the error)

haku15, 2019-05-03:
samsterckval, thank you very much for sharing your workaround.
I've quickly tested it on my device: the application does not crash, but I still see following error and the video capture does not work (image capture is ok). So I think  the problem is not yet solved.
```
File "_ctypes/callbacks.c", line 234, in 'calling callback function'
  File "/usr/lib/python3/dist-packages/picamera/mmalobj.py", line 1227, in wrapper
    self._pool.send_buffer(block=False)
TypeError: silent_send_buffer() missing 1 required positional argument: 'port'
```