google/aiyprojects-raspbian #587: No beep or light from Joy Demo on first or second startups

rayzrocket, 2019-03-09:
We hear no beep nor button light to indicate Joy Demo is working, on first start up nor subsequent start-ups.  Tried fresh SD card with only wall plug power and seperately tried with monitor, mouse, and keyboard.

Using new Google AIY G950-00866-01 kit with SD card and v2.1 camera from Mouser.  Checked flex connections 3 times.
Also tried update SD card image to 'aiyprojects-2018-11-16.img.xz'

We did the: ~ $ echo over_voltage=4 | sudo tee -a /boot/config.txt && sudo shutdown -r now

~ $ sudo systemctl status joy_detection_demo.service
 joy_detection_demo.service - AIY Joy Detection Demo
   Loaded: loaded (/home/pi/AIY-projects-python/src/examples/vision/joy/joy_detection_demo.service; enabled; vendor preset: enabled)
   Active: inactive (dead) since Sat 2019-03-09 16:03:31 GMT; 3min 22s ago
  Process: 468 ExecStart=/usr/bin/python3 /home/pi/AIY-projects-python/src/examples/vision/joy/joy_detection_demo.py --enable_streaming --mdn
 Main PID: 468 (code=exited, status=0/SUCCESS)

Mar 09 16:03:29 raspberrypi python3[468]:     self._init_preview()
Mar 09 16:03:29 raspberrypi python3[468]:   File "/usr/lib/python3/dist-packages/picamera/camera.py", line 513, in _init_preview
Mar 09 16:03:29 raspberrypi python3[468]:     self, self._camera.outputs[self.CAMERA_PREVIEW_PORT])
Mar 09 16:03:29 raspberrypi python3[468]:   File "/usr/lib/python3/dist-packages/picamera/renderers.py", line 558, in __init__
Mar 09 16:03:29 raspberrypi python3[468]:     self.renderer.inputs[0].connect(source).enable()
Mar 09 16:03:29 raspberrypi python3[468]:   File "/usr/lib/python3/dist-packages/picamera/mmalobj.py", line 2212, in enable
Mar 09 16:03:29 raspberrypi python3[468]:     prefix="Failed to enable connection")
Mar 09 16:03:29 raspberrypi python3[468]:   File "/usr/lib/python3/dist-packages/picamera/exc.py", line 184, in mmal_check
Mar 09 16:03:29 raspberrypi python3[468]:     raise PiCameraMMALError(status, prefix)
Mar 09 16:03:29 raspberrypi python3[468]: picamera.exc.PiCameraMMALError: Failed to enable connection: Out of resources

My 12yr old son is so excited to see this work eventually.
Please help or point me to a thread of similar issue.

miksch27, 2019-03-30:
same to me, After one sound , demo tells I haven't any sound installed..  I hope it was because of an sudo apt-get ..

manoj7410, 2020-05-12:
Is this issue reproducible with https://github.com/google/aiyprojects-raspbian/releases/tag/v20191113 ?

dmitriykovalev, 2020-11-23:
Please try the latest [SD card image](https://github.com/google/aiyprojects-raspbian/releases/tag/v20201120).

closed 2020-11-23 by dmitriykovalev