google/aiyprojects-raspbian #591: AIY vision examples are not working, in my Raspberry pi 3 B+

amitkray31, 2019-03-18:
AIY voice examples are working fine but the vision examples are not working. Can someone help me to solve this issue.
I am using pi camera v2.1 and my raspberry pi running aiyprojects-2018-11-16 OS

### pi@raspberrypi:~/AIY-projects-python/src/examples/vision $ python3 face_detection_camera.py
Traceback (most recent call last):
  File "face_detection_camera.py", line 78, in <module>
    main()
  File "face_detection_camera.py", line 63, in main
    with CameraInference(face_detection.model()) as inference:
  File "/opt/aiy/projects-python/src/aiy/vision/inference.py", line 100, in __init__
    self._engine = self._stack.enter_context(InferenceEngine())
  File "/opt/aiy/projects-python/src/aiy/vision/inference.py", line 274, in __init__
    self._transport = make_transport()
  File "/opt/aiy/projects-python/src/aiy/_drivers/_transport.py", line 87, in make_transport
    return _SpiTransport()
  File "/opt/aiy/projects-python/src/aiy/_drivers/_transport.py", line 29, in __init__
    self._spicomm = _spicomm.Spicomm()
  File "/opt/aiy/projects-python/src/aiy/_drivers/_spicomm.py", line 317, in __init__
    super().__init__()
  File "/opt/aiy/projects-python/src/aiy/_drivers/_spicomm.py", line 219, in __init__
    self._dev = os.open(SPICOMM_DEV, os.O_RDWR)
FileNotFoundError: [Errno 2] No such file or directory: '/dev/vision_spicomm'
### pi@raspberrypi:~/AIY-projects-python/src/examples/vision $ ls
any_model_camera.py     dish_detection.py       face_detection_camera.py  face_detection_raspivid.py      image_classification.py     $
dish_classification.py  face_camera_trigger.py  face_detection.py         image_classification_camera.py  inaturalist_classification.p$
### pi@raspberrypi:~/AIY-projects-python/src/examples/vision $ python3 image_classification_camera.py
Traceback (most recent call last):
  File "image_classification_camera.py", line 64, in <module>
    main()
  File "image_classification_camera.py", line 56, in main
    CameraInference(image_classification.model()) as inference:
  File "/opt/aiy/projects-python/src/aiy/vision/inference.py", line 100, in __init__
    self._engine = self._stack.enter_context(InferenceEngine())
  File "/opt/aiy/projects-python/src/aiy/vision/inference.py", line 274, in __init__
    self._transport = make_transport()
  File "/opt/aiy/projects-python/src/aiy/_drivers/_transport.py", line 87, in make_transport
    return _SpiTransport()
  File "/opt/aiy/projects-python/src/aiy/_drivers/_transport.py", line 29, in __init__
    self._spicomm = _spicomm.Spicomm()
  File "/opt/aiy/projects-python/src/aiy/_drivers/_spicomm.py", line 317, in __init__
    super().__init__()
  File "/opt/aiy/projects-python/src/aiy/_drivers/_spicomm.py", line 219, in __init__
    self._dev = os.open(SPICOMM_DEV, os.O_RDWR)
FileNotFoundError: [Errno 2] No such file or directory: '/dev/vision_spicomm'

closed 2019-03-19 by amitkray31

CGev, 2019-03-21:
Did you find a solution? I have the same probleme...

amitkray31, 2019-03-21:
> Did you find a solution? I have the same probleme...

Actually I thought that, I can run AIY-Vision example by just using the pi camera. Without adding any kind of hat( AIY-Vision bonet) just like AIY-voice examples(I skip the instruction so I ended up with those thoughts). Since I don't have AIY-vision bonnet I can't run AIY-vision examples.

**You can try the following steps. It may solve your problem.**
install  libttspico-utils
**`sudo apt-get -y install libttspico-utils`**

_**You must stack the AIY-vision bonnet over the raspberry pi to run the AIY-vision examples, if you don't it will show error**_