google/aiyprojects-raspbian #592: aiy.vision.leds missing Color class

kbrooks80, 2019-03-19:
When aiy.leds moved to aiy.vision.leds it looks like the Color class got stripped out and  __ enter __ and __ exit __ got stripped from the Leds class. Additionally, aiy.vision.leds does not import math anymore for the blend function in the Color class. I'm not sure if this was to increase performance or not - I'm still pretty terrible at Python & don't know how to do pull requests yet. The result of these omissions is that leds_example.py from the aiy.leds documentation does not work.

When you add the above items to aiy.vision.leds and change aiy.leds to aiy.vision.leds in leds_example.py, all examples in leds_example.py work.

kbrooks80, 2019-03-19:
This applies to the 2018-11-16 release image

dmitriykovalev, 2019-11-14:
There is no `aiy.vision.leds`, everything is in [`aiy.leds`](https://github.com/google/aiyprojects-raspbian/blob/aiyprojects/src/aiy/leds.py).

closed 2019-11-14 by dmitriykovalev