google/aiyprojects-raspbian #593: Error drivers: [Errno2] No such file or directory: '/dev/vision_spicomm'

CGev, 2019-03-21:
I wrote my own code: connect the raspberry to a websocket in order to take a photo and send it when it detects a face. I managed to run it but after few times I get the following error **[Errno2] No such file or directory: '/dev/vision_spicomm'** 
I tried to run joy_detection_demo.py but it doesn't work anymore and I have the same error. 
I tried "apt-get --reinstall install aiy-dkms"
I followed [HACKING.md 
](https://github.com/google/aiyprojects-raspbian/blob/v20181116/HACKING.md) and reinstall the lastest version of the SD card release
Neither work.
Does anyone have a solution?

hawksley71, 2019-04-29:
I get the same error message. Joy demo worked for about an hour, then tried the next demo (object detection) and I get this error. Tried running the Joy demo again, but it no longer works. The boards feel warm, but not hot.
Did you ever find a solution?

CGev, 2019-04-29:
Hi !
The only solution I found to this problem is to reinstall aiy vision
kernels and then reboot:
*sudo apt install --reinstall aiy-vision-dkms*
*sudo reboot*

Does it solve your problem?

hawksley71, 2019-04-30:
I reflashed my SD card..., but this time declined the updates to my Raspberry Pi after the restart. Now everything works. However, from your comments it looks like maybe I could have updated my Raspberry Pi and then followed your advice below. 
...will try that tomorrow.

georgikeith, 2019-10-15:
I'm seeing this error after a fresh install of the 2018-11-16 image, followed by an 'apt-get update; apt-get upgrade'.

reinstalling aiy-vision-dkms and rebooting did not work for me.

manoj7410, 2019-11-14:
This issue seems to be fixed in the latest release https://github.com/google/aiyprojects-raspbian/releases/tag/v20191113.

georgikeith, 2019-11-14:
Thanks!  I'll try ot out this weekend

George Keith
http://www.georgekeith.net

closed 2020-05-13 by dmitriykovalev