google/aiyprojects-raspbian #595: Vision Kit booting with red button light

ChilliCheeseKid, 2019-03-22:
Hey fellas
like it said in the title my Kit boots but it doenst make a sound or shows a green button. The button instead flashes red, nut even that doenst happen on every startup.
I allready tried the voltage override, checked the flex cables, flashed/changed the sd-card and used a wall socket.
Any other ideas? 
[image]
[image]
[image]

Best regards

CHKRISTINA, 2019-04-04:
I have the same problem. Did you solve it?

alexpazzo, 2019-04-05:
Hello, try to swith the direction of the cable that connect the raspberry and the gogole board.

ChilliCheeseKid, 2019-04-09:
Hey, thanks for the reply.
I tried it with a reversed calbe connection between bonnet and raspberry.
The only result was a very hight pitched constant noise.

mark10401, 2019-04-28:
I had a problem where the pi would not boot at all using  the short ribbon cable orientation as directed. In my case, I had to ensure that the labeling on the ribbon was facing me in order for the system to boot. This is exactly opposite from the instruction on step #28 at https://aiyprojects.withgoogle.com/vision/#assembly-guide  where it says _"Make sure the side with the copper stripes (and labels) is still facing away from you, as shown in the picture._" I'm not sure if this will solve all of your problems but I hope it helps.

bumatic, 2019-05-09:
I ran into the same problem. It turned out that my camera cable got loose somehow. Since identifying the error is the first step for fixing it, I thought it might be helpful to document how I figured out what's wrong:

On startup the RaspberryPi starts the `joy_detection_demo.service`. The three blinking red lights apparently indicate that the service crashed (they blink as well when the service is stopped manually). 

For figuring out whats wrong ssh into your RaspberryPi and run `sudo systemctl status joy_detection_demo.service`. 

In my case the output indicated that the camera is not running. If all cables are connected properly and the problem still occurs there might be an issue with the cable or the vision bonnet.

manoj7410, 2019-11-14:
@ChilliCheeseKid Can you share the output of "raspistill -o test.jpg" ?

Muffocean, 2020-08-18:
Hi, I have the same problem. Because nobody has come back since two years ago, I'm guessing some of you have found solutions. If this is still an open case, I'm wondering what "raspistill -o test.jpg" is.