google/aiyprojects-raspbian #600: mnist cnn on vision kit

byo91, 2019-03-31:
hi, it's possible to train and deploy a "homemade" cnn to run on the vision kit?
i have trained a model like this: https://github.com/tensorflow/docs/blob ... /cnn.ipynb

the bonnet compile return me this error:

2019-03-17 19:53:04.396130: I external/org_tensorflow/tensorflow/contrib/lite/toco/graph_transformations/graph_transformations.cc:39] Before Removing unused ops: 31 operators, 48 arrays (0 quantized)
2019-03-17 19:53:04.396533: I external/org_tensorflow/tensorflow/contrib/lite/toco/graph_transformations/graph_transformations.cc:39] Before general graph transformations: 31 operators, 48 arrays (0 quantized)
2019-03-17 19:53:04.396601: F external/org_tensorflow/tensorflow/contrib/lite/toco/graph_transformations/propagate_fixed_sizes.cc:347] Check failed: output_depth == input_depth * op->depth_multiplier (1 vs. 0)input/output depths and depth_multiplier don't match

what i can do?

does the vision kit only support the mobilenet model? is the only way to retrain the mobilenet and transfer learning?

thanks

manoj7410, 2019-11-14:
Please see the detailed documentation at https://aiyprojects.withgoogle.com/vision/#makers-guide--tensorflow-model-compiler regarding the limitations that bonnet_model_compiler has.

zye1996, 2020-02-21:
> hi, it's possible to train and deploy a "homemade" cnn to run on the vision kit?
> i have trained a model like this: https://github.com/tensorflow/docs/blob ... /cnn.ipynb
>
> the bonnet compile return [...]

Have you solved the problem? Exactly the same problem here