google/aiyprojects-raspbian #608: apt's aiyprojects repo key troubles

themaire, 2019-04-18:
Hi,

I am a AIY V1 Raspbian user (last image) user. 
Since few days, "apt update" command have a key error with the google aiyproject repository. A key error.

`In French sorry...

pi@aiy:~ $ sudo apt update
Atteint:1 http://archive.raspberrypi.org/debian stretch InRelease
Atteint:2 http://raspbian.raspberrypi.org/raspbian stretch InRelease
Ign:3 https://dl.google.com/aiyprojects/deb stable InRelease   
Atteint:4 https://dl.google.com/aiyprojects/deb stable Release
Atteint:5 https://deb.nodesource.com/node_8.x stretch InRelease
Err:6 https://dl.google.com/aiyprojects/deb stable Release.gpg
  Les signatures suivantes ne sont pas valables : EXPKEYSIG 1397BC53640DB551 Google Inc. (Linux Packages Signing Authority) <linux-packages-keymaster@google.com>
Lecture des listes de paquets... Fait
Construction de l'arbre des dépendances       
Lecture des informations d'état... Fait
5 packages can be upgraded. Run 'apt list --upgradable' to see them.
W: An error occurred during the signature verification. The repository is not updated and the previous index files will be used. GPG error: https://dl.google.com/aiyprojects/deb stable Release: Les signatures suivantes ne sont pas valables : EXPKEYSIG 1397BC53640DB551 Google Inc. (Linux Packages Signing Authority) <linux-packages-keymaster@google.com>
W: Impossible de récupérer https://dl.google.com/aiyprojects/deb/dists/stable/Release.gpg  Les signatures suivantes ne sont pas valables : EXPKEYSIG 1397BC53640DB551 Google Inc. (Linux Packages Signing Authority) <linux-packages-keymaster@google.com>
W: Le téléchargement de quelques fichiers d'index a échoué, ils ont été ignorés, ou les anciens ont été utilisés à la place.`

I tryed to add/update the right key with this command, no succes ([https://www.google.com/linuxrepositories/](url)) :

Anyone have a idea to fix the problem please? I can't upgrade my package with this apt trouble. Thanks!

themaire, 2019-04-23:
Ok, the problem is no longer relevant because the command "apt update" has no more error about the repo """  https://dl.google.com/aiyprojects/deb  """.

Good news! Issue resolved.

mattyyey, 2019-05-01:
I had the same issue. I commented out the line in /etc/apt/sources.list.d/aitprojects.list
That bypassed the problem. Now I have uncommented it nearly a month later, after seeing this it works.

dmitriykovalev, 2019-11-14:
Apt repo URL was changed in the latest SD card image, please check [HACKING.md](https://github.com/google/aiyprojects-raspbian/blob/aiyprojects/HACKING.md) for details.

closed 2019-11-14 by dmitriykovalev