google/aiyprojects-raspbian #609: Source and Destination checksums do not match

chiragbhagat, 2019-04-21:
I got the following error while using Etcher to flash the .img.xz file onto my MicroSD card.

\\.\PhysicalDrive1: Source and Destination checksums do not match: a36f58668eebd042!==44c89104a42cf590

Please help!

Thanks,
Chirag

wertyy102, 2019-04-27:
@chiragbhagat have you extracted the *.img.xz using the command 
`unxz ./aiyprojects-20yy-mm-dd.img.xz 
` 
and then tried using dd to write the .img file to an sd card?

etchr works great 99.999% of the time, until it comes to debugging things like this.

if you need some help with dd, it works like this

`bash
#dd if=/path/to/image/file.img of=/dev/sdX`
where X is the physical device name, found using the command 
`bash
# fdisk -l 
` from a root shell.  Hope this helps!

dmitriykovalev, 2019-11-14:
I guess you can try to flash one more time if checksum check fails. If it's consistent then try to use different SD card.

closed 2019-11-14 by dmitriykovalev