google/aiyprojects-raspbian #614: Failed booting. The button is flashing with a red light.

Gago55, 2019-05-07:
Hi guys, I followed the instruction, build Vision, install the last image on sdCard, connect the camera to power with power supply(5V - 2A). After 10 minutes without any sound, the button flashing with a red light. I tried anything, reflash sdCard a few times, even I tried to flash sdCard with previous version 2018-08-03 but nothing changes. I'll glad for any help.

jamesallen9, 2019-05-07:
Hi, I know what’s the problem. The same issue was happened with me. Do you have whatsapp or viber, then I can communicate to you?

Gago55, 2019-05-07:
> Hi, I know what’s the problem. The same issue was happened with me. Do you have whatsapp or viber, then I can communicate to you?
Yes man I have viber , can you give your mail , I will send you my contacts

jamesallen9, 2019-05-07:
Yes here is my email address 
allenjames716@yahoo.com

Gago55, 2019-05-07:
Man but I think it is better to tell the solution here so anybody with the same issue can fix it.

jamesallen9, 2019-05-07:
Ok, how do you want, but I think it will be easy to chat with you, or we can have a videocall with skype to tell you what to do )

jamesallen9, 2019-05-07:
I create github account special for your issue )

Gago55, 2019-05-07:
Anyway  it will be better to tell the solution here if it is not difficult

jamesallen9, 2019-05-07:
🖕🏿🖕🏿

NagaKanaparthy, 2019-08-11:
Could you provide pictures of the kit's internal setup. I think it may be a wire being loose or not fully connected. Most likely those yellow flex connectors.

Gago55, 2019-08-11:
With connections all is OK. Problem in camera, it is defective. I've tried with my friend's camera and all was working.

NagaKanaparthy, 2019-08-11:
Since this was a hardware issue (camera was defective as a Dead On Arrival part), I don't think this issue should stay open. Could you please close the issue?

closed 2019-08-11 by Gago55