google/aiyprojects-raspbian #616: AIY voice kit V1.0 - need help

boxshifter, 2019-05-09:
Hi

When i try to do the setup credentials on google cloud platform (OAuth consent screen) i'm seeing a totally different screen to the one in the setup guide, instead of being asked for an email and product name like the guide shows i'm being asked for Application Name, Application home page link, Application logo, Support email, Authorised domains etc.... and don't have a clue how to get past his?

manoj7410, 2019-11-14:
@boxshifter what do you see on the credentials tab ?

manoj7410, 2020-05-12:
You only need to provide your App name (Sample : Voice kit) and nothing else is required on the consent screen.

closed 2020-05-13 by dmitriykovalev

schamu78, 2020-06-07:
I follow the steps of providing the App name like you suggested but then I get an error that says "the request failed because one of the field of the resource is invalid."

djwdev, 2020-06-10:
I got the same error "the request failed because one of the field of the resource is invalid".  Does Anybody have the work around?

schamu78, 2020-06-10:
I reported to the support email and this is what I got.
[image]

manoj7410, 2020-06-12:
I followed the below procedure and it worked fine for my personal account:

1. Create a new project.
2. Select the project that just got created.
3. Goto APIs and Services and Click on Library
4. Search "assistant" and Select "Google Assistant API"
5. Enable the API
6. Click on "Create Credentials".
7. Select "Google Assistant API" in "Which API are you using?" option.
8. Select "Other UI" in "Where will you be calling the API from".
9. Select "User Data" and click "What credentials do I need?".
10. Click on "SET UP CONSENT SCREEN".
11. Select "External" in "User Type".
12. Enter an application name on the consent screen page and no other information needs to be provided.
13. Click "Save" on Consent Screen.
14. Click on "Credentials" again. 
15. Click on "Create Credentials" on top.
16. Select OAuth Client ID
17. Select "Desktop App" and Enter client name and Click on Create.
18. Download the credentials and use them on voice kit with old steps.

Attached are all the screenshots with step number.

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]