google/aiyprojects-raspbian #617: Voice Kit v2, and  a "aplay: playback:2787" with the Mic...

SalvorinFex, 2019-05-09:
I just installed the latest AIY SD card image on my v2 voice kit but the Mic fails with this error:
`aplay: playback:2787; read error`

I tried updating the image via `apt update and upgrade` but then it found no sound card at all, so I re-imaged with the 11-2018 image.

What is going on here and how can I fix this?

SalvorinFex, 2019-05-17:
Can anyone please chime in here on this issue?
I am unable to proceed to use my Kit due to this.

rickyklu, 2019-05-28:
There have been issues with running `apt-get update` and `apt-get upgrade` breaking the AIY voice kit functionality. It seems to be an issue with the fact that the latest kernels do not work well with the AIY hardware. But it doesn't look like there is recent official development going on that will fix it. You can try the solution [here](https://github.com/google/aiyprojects-raspbian/issues/527#issuecomment-455791055), otherwise re-installing the image and don't run `update` or `upgrade`

andrewtvuong, 2019-07-04:
agreed with @rickyklu

manoj7410, 2019-11-14:
@SalvorinFex Can you try to reflash your kit with https://github.com/google/aiyprojects-raspbian/releases/tag/v20191113 and see if the issue still persists ?

dmitriykovalev, 2020-05-13:
Please reopen this one if the issue still exists on the latest SD card image.

closed 2020-05-13 by dmitriykovalev