google/aiyprojects-raspbian #618: AIY Vision Kit: Could not enable camera whatsoever

tommyjtl, 2019-05-10:
Just assembled my AIY Vision Kit (only hardware part), at startup it won't run the default Joy Detection demo, then I stopped the daemon service, and try the image classification demo from inside AIY-projects-python folder (`~/AIY-projects-python/src/examples/vision/`), then it shows:

```shell
pi@raspberrypi:~/AIY-projects-python/src/examples/vision $ ./image_classification_camera.py 
mmal: mmal_vc_port_enable: failed to enable port vc.null_sink:in:0(OPQV): ENOSPC
mmal: mmal_port_enable: failed to enable connected port (vc.null_sink:in:0(OPQV))0x1b06880 (ENOSPC)
mmal: mmal_connection_enable: output port couldn't be enabled
Traceback (most recent call last):
  File "./image_classification_camera.py", line 64, in <module>
    main()
  File "./image_classification_camera.py", line 54, in main
    with PiCamera(sensor_mode=4, framerate=30) as camera, \
  File "/usr/lib/python3/dist-packages/picamera/camera.py", line 433, in __init__
    self._init_preview()
  File "/usr/lib/python3/dist-packages/picamera/camera.py", line 513, in _init_preview
    self, self._camera.outputs[self.CAMERA_PREVIEW_PORT])
  File "/usr/lib/python3/dist-packages/picamera/renderers.py", line 558, in __init__
    self.renderer.inputs[0].connect(source).enable()
  File "/usr/lib/python3/dist-packages/picamera/mmalobj.py", line 2212, in enable
    prefix="Failed to enable connection")
  File "/usr/lib/python3/dist-packages/picamera/exc.py", line 184, in mmal_check
    raise PiCameraMMALError(status, prefix)
picamera.exc.PiCameraMMALError: Failed to enable connection: Out of resources
```

it doesn't works, then I ran `raspistill` to check if this util could work:

```shell
pi@raspberrypi:~/AIY-projects-python/src/examples/vision $ raspistill -o output.jpg
mmal: mmal_vc_component_enable: failed to enable component: ENOSPC
mmal: camera component couldn't be enabled
mmal: main: Failed to create camera component
mmal: Failed to run camera app. Please check for firmware updates
```

It did not work. Then I did some upgrade things like here https://github.com/google/aiyprojects-raspbian/issues/390:

```shell
sudo apt-get update
sudo apt-get upgrade
sudo rpi-update
```

After all upgrades done, the demo still not working.

I have checked my wiring, no problem at all (at least I think so), here's what the whole thing look like:

[image]

[image]

I have try to tighten the short flex between boards, but again, not working whatsoever. Can any one help me fix this?

MrSolrac, 2019-05-22:
Have you re-flashed you image? I had an issue with mine that it would not work once I did an update to the stock image . Similar to this issue https://github.com/google/aiyprojects-raspbian/issues/505?

MrSolrac, 2019-05-22:
Also, may look at https://github.com/google/aiyprojects-raspbian/issues/527

simonfrfr, 2019-09-04:
I purchased two and one works but the other does not... I just swapped the SD card from one into the other, and I get the same error on one system. I think it might be a hardware issue with your device, I'll let you know if I find anything.

manoj7410, 2019-11-14:
@tommyjtl Can you try to reflash the kit with https://github.com/google/aiyprojects-raspbian/releases/tag/v20191113 ? 
If the issue still persists, try to connect the camera directly to your PI and run "raspistill -o test.jpg" and share the output.

ricardodeazambuja, 2020-02-04:
I'm also having problems with the camera (original, V2.1) when it's connected through the AIY vision hat. When I connect it directly to the RPI Zero W, I'm able to run "raspistill -o test.jpg". The same command when connected to the AIY hat gives this msg:
```Camera control callback  cmd=0x4f525245mmal: No data received from sensor. Check all connections, including the Sunny one on the camera board```

BTW, I just flashed the sdcard using the "aiyprojects-2019-11-13.img" after I got this problem. The funny thing is that I was using the camera connected to the AIY hat, around two months ago, without problems. What is the image doing during the first boot? I can see it's expanding the filesystem, is it also updating something?

ricardodeazambuja, 2020-02-05:
Ok, following my tests, I flashed, again, the image recommended by @manoj7410 and this time I run ```sudo apt upgrade```, you know, just in case the problem was solved in the latest version. After that, raspistill behaviour changed!
$ raspistill -v
```
"raspistill" Camera App (commit a14ee020d361 Tainted)

Camera Name imx219
Width 3280, Height 2464, filename (null)
Using camera 0, sensor mode 0

GPS output Disabled

Quality 85, Raw no
Thumbnail enabled Yes, width 64, height 48, quality 35
Time delay 5000, Timelapse 0
Link to latest frame enabled  no
Full resolution preview No
Capture method : Single capture

Preview Yes, Full screen Yes
Preview window 0,0,1024,768
Opacity 255
Sharpness 0, Contrast 0, Brightness 50
Saturation 0, ISO 0, Video Stabilisation No, Exposure compensation 0
Exposure Mode 'auto', AWB Mode 'auto', Image Effect 'none'
Flicker Avoid Mode 'off'
Metering Mode 'average', Colour Effect Enabled No with U = 128, V = 128
Rotation 0, hflip No, vflip No
ROI x 0.000000, y 0.000000, w 1.000000 h 1.000000
mmal: mmal_vc_component_enable: failed to enable component: ENOSPC
mmal: camera component couldn't be enabled
mmal: main: Failed to create camera component
mmal: Failed to run camera app. Please check for firmware updates
```

And, apparently, the firmware changed too:
$ /opt/vc/bin/vcgencmd version
```
Jan 10 2020 17:02:31 
Copyright (c) 2012 Broadcom
version f3e39fed4763ce4efbffabb2d4787aaacddeda0f (clean) (release) (start_x)
```

However, I'm still capable to use the AIY hat, e.g. image_classification.py, on a previously saved image.

tuttelikz, 2020-06-18:
`raspivid -t 10000` was not working initially.
But once I did recommended:
`sudo apt-get update`
`sudo apt-get upgrade` # took long time
`sudo rpi-update`
my camera started to display stream from raspivid

ricardodeazambuja, 2020-06-22:
@tuttelikz Would you mind telling me if your camera was connected to the AIY Vision Bonnet or directly to the RPI Zero W?

tuttelikz, 2020-06-23:
> @tuttelikz Would you mind telling me if your camera was connected to the AIY Vision Bonnet or directly to the RPI Zero W?

@ricardodeazambuja Through Bonnet, omg, forgot to mention that.

ricardodeazambuja, 2020-06-23:
@tuttelikz Thanks for your answer!
Now, a probably silly question: is everything working on the AIY Vision Bonnet after the upgrade and which raspbian version are you using?

tuttelikz, 2020-06-24:
What the irony, @ricardodeazambuja. I tried to `raspivid` right after you asked, now it gives exactly same error as yours:
`Camera control callback cmd=0x4f525245mmal: No data received from sensor. Check all connections, including the Sunny one on the camera board`

As for question asked: Yes, it WAS working on RPI Zero W, **aiyprojects-raspbian**, [AIY Kits Release 2019-11-13](https://github.com/google/aiyprojects-raspbian/releases), connecting Raspberry Camera Module v2 via AIY Vision Bonnet.
Here is the response to `/opt/vc/bin/vcgencmd version`:
```
Jun 19 2020 15:45:52
Copyright (c) 2012 Broadcom
version 31eda665dcc45e61650a2ef018ef2cf94091c46f (clean) (release) (start_x)
```
Hope we find a clue how solve it.

ricardodeazambuja, 2020-06-24:
@tuttelikz, my solution was to connect the camera directly to the RPI Zero W. Everything works fine that way, but I would like to verify if it's much slower compared with the direct connection to the AIY Vision Bonnet. So far I think the speed is the same or almost the same.

shoulih, 2022-08-23:
with the same Vision Kit long flex cable?