google/aiyprojects-raspbian #619: Can the voice bonnet be used with Ubuntu?

ChronoSynth, 2019-05-10:
I am building a social robot with Raspberry Pi. I currently have Ubuntu MATE installed on it. I want to use the AIY voice bonnet as a microphone, but I don't want to switch OS's because my current OS has ROS installed on it. Is it possible to use the voice bonnet on an Ubuntu OS? If so, how?

aniquetahir, 2019-06-06:
Yes you can. Check out the instructions here:
https://github.com/google/aiyprojects-raspbian/blob/aiyprojects/HACKING.md

closed 2019-11-14 by dmitriykovalev

anton-matosov, 2022-05-06:
I have tried following Hacking.md with Ubuntu 22.04 server on Raspi 3B and it ends with
```
pi@dr-qp ~> sudo apt-get install -y leds-ktd202x-dkms pwm-soft-dkms aiy-usb-gadget aiy-bt-prov-server aiy-vision-dkms aiy-models aiy-python-wheels
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 aiy-vision-dkms : Depends: raspberrypi-kernel-headers but it is not installable
                   Depends: aiy-dkms but it is not going to be installed
 leds-ktd202x-dkms : Depends: raspberrypi-kernel-headers but it is not installable
 pwm-soft-dkms : Depends: raspberrypi-kernel-headers but it is not installable
E: Unable to correct problems, you have held broken packages.
```
So the current answer is hard NO

----
For getting the AIY Vision + ROS 2 support I am going to try installing 64 bit Raspbian and running ROS in docker container

...if that fails
The next step would be to either switch to `Intel® Neural Compute Stick 2` or use `NVIDIA Jetson`. The latter has direct support for robotics via ROS, resulting in less hacking and more doing (pun intended)

anton-matosov, 2022-05-06:
Rasperry PI OS 64 bit fails as well
```
sudo apt-get install -y leds-ktd202x-dkms pwm-soft-dkms aiy-usb-gadget aiy-bt-prov-server aiy-vision-dkms aiy-models aiy-python-wheels
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
aiy-bt-prov-server is already the newest version (1.2-1).
aiy-models is already the newest version (1.1-0).
aiy-python-wheels is already the newest version (1.4-1).
aiy-usb-gadget is already the newest version (1.0-1).
aiy-vision-dkms is already the newest version (1.2-1).
leds-ktd202x-dkms is already the newest version (1.2-1).
pwm-soft-dkms is already the newest version (2.0-1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
4 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
Setting up aiy-python-wheels (1.4-1) ...
WARNING: Requirement '/opt/aiy/python-wheels/protobuf-3.14.0-cp37-cp37m-linux_aarch64.whl' looks like a filename, but the file does not exist
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
ERROR: protobuf-3.14.0-cp37-cp37m-linux_aarch64.whl is not a supported wheel on this platform.
dpkg: error processing package aiy-python-wheels (--configure):
 installed aiy-python-wheels package post-installation script subprocess returned error exit status 1
dpkg: dependency problems prevent configuration of aiy-bt-prov-server:
 aiy-bt-prov-server depends on aiy-python-wheels; however:
  Package aiy-python-wheels is not configured yet.

dpkg: error processing package aiy-bt-prov-server (--configure):
 dependency problems - leaving unconfigured
Setting up aiy-dkms (2.0-1) ...
Removing old aiy-2.0 DKMS files...

-------- Uninstall Beginning --------
Module:  aiy
Version: 2.0
Kernel:  5.10.0-14-arm64 (aarch64)
-------------------------------------

Status: Before uninstall, this module version was ACTIVE on this kernel.

aiy-io-i2c.ko:
 - Uninstallation
   - Deleting from: /lib/modules/5.10.0-14-arm64/updates/dkms/
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.

pwm-aiy-io.ko:
 - Uninstallation
   - Deleting from: /lib/modules/5.10.0-14-arm64/updates/dkms/
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.

gpio-aiy-io.ko:
 - Uninstallation
   - Deleting from: /lib/modules/5.10.0-14-arm64/updates/dkms/
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.

aiy-adc.ko:
 - Uninstallation
   - Deleting from: /lib/modules/5.10.0-14-arm64/updates/dkms/
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.

depmod...

DKMS: uninstall completed.

------------------------------
Deleting module version: 2.0
completely from the DKMS tree.
------------------------------
Done.
Loading new aiy-2.0 DKMS files...
It is likely that 5.15.32-v8+ belongs to a chroot's host
Building for 5.10.0-14-arm64 and 5.15.32-v8+
Building initial module for 5.10.0-14-arm64
Done.

aiy-io-i2c.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.10.0-14-arm64/updates/dkms/

pwm-aiy-io.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.10.0-14-arm64/updates/dkms/

gpio-aiy-io.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.10.0-14-arm64/updates/dkms/

aiy-adc.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.10.0-14-arm64/updates/dkms/

depmod...

DKMS: install completed.
Building initial module for 5.15.32-v8+
Error! Bad return status for module build on kernel: 5.15.32-v8+ (aarch64)
Consult /var/lib/dkms/aiy/2.0/build/make.log for more information.
dpkg: error processing package aiy-dkms (--configure):
 installed aiy-dkms package post-installation script subprocess returned error exit status 10
dpkg: dependency problems prevent configuration of aiy-vision-dkms:
 aiy-vision-dkms depends on aiy-dkms; however:
  Package aiy-dkms is not configured yet.

dpkg: error processing package aiy-vision-dkms (--configure):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 aiy-python-wheels
 aiy-bt-prov-server
 aiy-dkms
 aiy-vision-dkms
E: Sub-process /usr/bin/dpkg returned an error code (1)

`