google/aiyprojects-raspbian #620: Internal Server Error - Repeat Issue

CarlP66, 2019-05-12:
Need to re-open issue #577 as per comment by  dmitriykovalev.
Repeat occurrence that I can't get past.

AIY Voice has been working fine for months. No changes on my side, but today when I run the python script that I have been using, I get:

"requests.exceptions.HTTPError: 500 Server Error: Internal Server Error for url: https://embeddedassistant.googleapis.com/v1alpha2/projects/voice-assistant-213123/deviceModels/voice-assistant-213123-voice-kit"

Please re-open & investigate.

Thanks, Carl.

manoj7410, 2019-11-14:
@CarlP66 Is the issue still reproducible ?

CarlP66, 2019-11-14:
No. The problem resolved itself as mysteriously as it occurred. I am pretty sure it was a transient issue at Google's end. I suspect I will see it again at some point......

dmitriykovalev, 2019-11-14:
@CarlP66 Sounds good, please reopen if it happens again.

closed 2019-11-14 by dmitriykovalev