google/aiyprojects-raspbian #627: Error! Bad return status for module build on kernel: 4.19.42+ (armv6l)

VinQbator, 2019-05-28:
Just got the AIY-vision kit and it worked fine at first. For some reason I did apt update and apt upgrade and after that it has been a long day of debugging, but nothing helps.

When trying to run any vision scripts I'm getting 
```
File "/home/pi/AIY-projects-python/src/aiy/_drivers/_spicomm.py", line 219, in __init__
    self._dev = os.open(SPICOMM_DEV, os.O_RDWR)
FileNotFoundError: [Errno 2] No such file or directory: '/dev/vision_spicomm'
```
I've uninstalled all aiy-* packages and reinstalled according to HACKING.md but still the same issue.

I think the kernel upgrade messed up the system. Is there any other way besides flashing the SD card?

When trying to install aiy-vision-dkms, I'm getting the following error.
```
DKMS make.log for aiy-1.1 for kernel 4.19.42+ (armv6l)
Tue May 28 19:28:33 UTC 2019
make: Entering directory '/usr/src/linux-headers-4.19.42+'
  AR      /var/lib/dkms/aiy/1.1/build/aiy/gpio/built-in.a
  CC [M]  /var/lib/dkms/aiy/1.1/build/aiy/gpio/gpio-aiy-io.o
  AR      /var/lib/dkms/aiy/1.1/build/aiy/iio/adc/built-in.a
  CC [M]  /var/lib/dkms/aiy/1.1/build/aiy/iio/adc/aiy-adc.o
/var/lib/dkms/aiy/1.1/build/aiy/iio/adc/aiy-adc.c:138:2: error: unknown field 'driver_module' specified in initializer
  .driver_module = THIS_MODULE,
  ^
In file included from ./include/linux/linkage.h:7:0,
                 from ./include/linux/kernel.h:7,
                 from ./include/asm-generic/gpio.h:5,
                 from ./arch/arm/include/asm/gpio.h:10,
                 from ./include/linux/gpio.h:59,
                 from /var/lib/dkms/aiy/1.1/build/aiy/iio/adc/../../include/aiy-io.h:20,
                 from /var/lib/dkms/aiy/1.1/build/aiy/iio/adc/aiy-adc.c:16:
./include/linux/export.h:16:21: error: initialization from incompatible pointer type [-Werror=incompatible-pointer-types]
 #define THIS_MODULE (&__this_module)
                     ^
/var/lib/dkms/aiy/1.1/build/aiy/iio/adc/aiy-adc.c:138:19: note: in expansion of macro 'THIS_MODULE'
  .driver_module = THIS_MODULE,
                   ^~~~~~~~~~~
./include/linux/export.h:16:21: note: (near initialization for 'aiy_adc_info.read_raw_multi')
 #define THIS_MODULE (&__this_module)
                     ^
/var/lib/dkms/aiy/1.1/build/aiy/iio/adc/aiy-adc.c:138:19: note: in expansion of macro 'THIS_MODULE'
  .driver_module = THIS_MODULE,
                   ^~~~~~~~~~~
cc1: some warnings being treated as errors
scripts/Makefile.build:309: recipe for target '/var/lib/dkms/aiy/1.1/build/aiy/iio/adc/aiy-adc.o' failed
make[4]: *** [/var/lib/dkms/aiy/1.1/build/aiy/iio/adc/aiy-adc.o] Error 1
scripts/Makefile.build:544: recipe for target '/var/lib/dkms/aiy/1.1/build/aiy/iio/adc' failed
make[3]: *** [/var/lib/dkms/aiy/1.1/build/aiy/iio/adc] Error 2
scripts/Makefile.build:544: recipe for target '/var/lib/dkms/aiy/1.1/build/aiy/iio' failed
make[2]: *** [/var/lib/dkms/aiy/1.1/build/aiy/iio] Error 2
scripts/Makefile.build:544: recipe for target '/var/lib/dkms/aiy/1.1/build/aiy' failed
make[1]: *** [/var/lib/dkms/aiy/1.1/build/aiy] Error 2
Makefile:1524: recipe for target '_module_/var/lib/dkms/aiy/1.1/build' failed
make: *** [_module_/var/lib/dkms/aiy/1.1/build] Error 2
make: Leaving directory '/usr/src/linux-headers-4.19.42+'
```

yoni2356, 2019-06-04:
I too cannot get vision bonnet to work on RPI strech lite (updated os and kernel, rpi zero)
I believe HACKING.md needs refactoring

regarding your specific issue, you can build aiy-vision-dkms if you run:
sudo apt-get --purge remove linux-headers-3.6-trunk-common

-> according to: https://stackoverflow.com/questions/54450251/build-of-aiy-vision-dkms-fails

personally I've tried and successfully built aiy-vision-dkms, yet still cannot get "Myriad ready" in dmesg

TheSwert, 2019-06-08:
I had a similar issue with my Voice kit. I found a script that will revert the kernel to 4.14.98 (last version I had installed that worked). After reverting the kernel version and rebooting you should be able to either do a dpkg-reconfigure on the installed modules, or just reinstall them.

https://gist.github.com/psyke83/cb3ca50561480809c246f42727cb7cf2

tarzanek, 2019-06-14:
after rpi kernel upgrade just rebuild of dkms packages is enough for me to get AIY back working
(see workaround in https://github.com/google/aiyprojects-raspbian/issues/570 )

itsalljustaride, 2019-06-16:
This project is in real need of a new disk image with updated kernel, etc. It has been nearly 8 months since the last working disk image release. Is the AIY project dead?

tarzanek, 2019-06-25:
had a quick look and the patches are not that hard to do
basically for 
 /var/lib/dkms/aiy/1.1/
apply https://github.com/raspberrypi/linux/commit/97623c0a80a605ef3fae337081ed008796bf8cc2#diff-3ec871a8e75c9ea85161805bbdee22a1 and https://github.com/raspberrypi/linux/commit/3f3a89e1d7c31558c070692241e3d6146d2cf1bf

the harder part is to update google deb repos for everybody to get the fixes

ginkage, 2019-07-23:
I got it working with kernel 4.19.57.

There are two files you'll need to edit (note that you'll need to use `sudo` to edit them). First one:
`/usr/src/aiy-1.1/aiy/iio/adc/aiy-adc.c`
(note that `/usr/src/aiy-1.1` is where the source code is copied from when the module is being built).

Remove the line starting with `.driver_module =`.

Second one:
`/usr/src/aiy-1.1/aiy/mfd/aiy-io-i2c.c`

Replace two occurrences (in `aiy_io_reset` and in `aiy_io_update_firmware`) of
`i2c_lock_adapter(i2c->adapter);`
with
`i2c_lock_bus(i2c->adapter, I2C_LOCK_ROOT_ADAPTER);`

and replace two occurrences of
`i2c_unlock_adapter(i2c->adapter);`
with
`i2c_unlock_bus(i2c->adapter, I2C_LOCK_ROOT_ADAPTER);`

Then run the following command:
`sudo dkms install aiy/1.1`

Reboot, and Bob's your uncle. Literally five lines of code to change. :)

tarzanek, 2019-07-27:
@ginkage will you be able to update deb packages too? ;-) 
Or know someone who can patch them and build and upload to googles repo?

CEBasile, 2019-07-30:
I have updated and rebuilt the package. You can get it here:
https://drive.google.com/open?id=1s6ru6Clz0pmX0VS3BOW9XJoqKE7E8y12

HeatfanJohn, 2019-07-31:
> I have updated and rebuilt the package. You can get it here:
> https://drive.google.com/open?id=1s6ru6Clz0pmX0VS3BOW9XJoqKE7E8y12

@CEBasile where is the source code all of the AIY debian packages?  It's not in this project, right?

CEBasile, 2019-08-01:
@HeatfanJohn This is just the python framework. The drivers are another repo. Look at HACKING.md

HeatfanJohn, 2019-08-05:
@CEBasile I see the reference to add the AIY vision package repo and I downloaded https://dl.google.com/aiyprojects/deb/dists/stable/Release to see the Debian release information:

```
Origin: Google LLC
Label: Google
Suite: stable
Codename: stable
Version: 1.0
Date: Fri, 19 Apr 2019 18:38:23 UTC
Architectures: armhf
Components: main
Description: Google AIY software repository
MD5Sum:
 7206381e7c543d5beb3272c52f6dff97 9183 main/binary-armhf/Packages
 7e0f0653397e626c7840c148722ea824 2840 main/binary-armhf/Packages.gz
 a0001a8b1dbdcd88b01d07009f8c3d8f 142 main/binary-armhf/Release
SHA1:
 01e66fff79dd958f5924f9d8057b16d1af9c3690 9183 main/binary-armhf/Packages
 83e85ca2e0f73c88c912eae473c6bfba22b3e726 2840 main/binary-armhf/Packages.gz
 bd6f1274a87b14ffcd4b110e1a3dc0cb60cf38eb 142 main/binary-armhf/Release
SHA256:
 5a01c5d0b68d0480c24daeafa42758e49eaafe6a009eebf4a4dd094caaf2fb23 9183 main/binary-armhf/Packages
 91a11fca702324b79486edef9cefe0b53e62c0eee69dfed7f0207e3a5ca4f913 2840 main/binary-armhf/Packages.gz
 5e600eb82427bc30bf6b1abdca202eb0413c1e41247d1891ef57a4c23bc55806 142 main/binary-armhf/Release
```
And then downloaded the package information https://dl.google.com/aiyprojects/deb/dists/stable/main/binary-armhf/Packages

But that only shows the `deb` locations.  Where / how do you find the source code?  Or is the source code included in the `deb` files?

sresam89, 2019-08-10:
> sudo dkms install aiy/1.1

Tried doing that 
`Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make -j4 KERNELRELEASE=4.19.58-v7l+ -C /lib/modules/4.19.58-v7l+/build M=/var/lib/dkms/aiy/1.1/build....(bad exit status: 2)
Error! Bad return status for module build on kernel: 4.19.58-v7l+ (armv7l)
Consult /var/lib/dkms/aiy/1.1/build/make.log for more information.
`

no go there

CEBasile, 2019-10-07:
Just wanted to let everyone know that downgrading to kernel 4.7 fixes these issues. Currently working on fixing this for kernel 4.19.66