google/aiyprojects-raspbian #630: Project abandoned?

surak, 2019-06-22:
There's no activity for about half a year... Is this going to Google Graveyard?

andrewtvuong, 2019-07-04:
@surak I had read at least one more update to the image was in the works, but it was maybe 6 months ago. Would Google completely throw away their work in the middle and not have a clean end?

How wonderful to be able to work on this team, I'd love for the opportunity.

jatpdubya, 2019-08-30:
[https://github.com/google/aiyprojects-raspbian/issues/635#issuecomment-524366391](url)
There are still enough people interested in using the voice kit finding workarounds for most of the issues. This comment is a good place to start if you're interested in using the hardware without having to use a dated custom image from google.

manoj7410, 2019-11-14:
There is another release available of AIY which can be found at : https://github.com/google/aiyprojects-raspbian/releases/tag/v20191113.

PeterMalkin, 2019-12-12:
As you can see the team continues to support this effort. If this will be scheduled to sunset, we will notify you on the github as well as on the website.

closed 2019-12-12 by PeterMalkin