google/aiyprojects-raspbian #631: Error installing aiy-python-wheels for AIY Voice (V1)

jsavage, 2019-06-27:
Hi,  I am trying to set up AIY Voice (v1) on Raspbian Buster following Hacking.md
 
When I run install -y aiy-python-wheels I see the following output and Error message.

I would be grateful for any pointers on what could be wrong.

Reading package lists... Done
Building dependency tree
Reading state information... Done
The following NEW packages will be installed:
  aiy-python-wheels
0 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
Need to get 18.2 MB of archives.
After this operation, 21.4 MB of additional disk space will be used.
Get:1 https://dl.google.com/aiyprojects/deb stable/main armhf aiy-python-wheels                                    all 1.1-1 [18.2 MB]
Fetched 18.2 MB in 4s (4,718 kB/s)
Selecting previously unselected package aiy-python-wheels.
(Reading database ... 91978 files and directories currently installed.)
Preparing to unpack .../aiy-python-wheels_1.1-1_all.deb ...
Unpacking aiy-python-wheels (1.1-1) ...
Setting up aiy-python-wheels (1.1-1) ...
protobuf-3.6.1-cp35-cp35m-linux_armv6l.whl is not a supported wheel on this plat                                   form.
dpkg: error processing package aiy-python-wheels (--configure):
 installed aiy-python-wheels package post-installation script subprocess returne                              d error exit status 1
Errors were encountered while processing:
 aiy-python-wheels
E: Sub-process /usr/bin/dpkg returned an error code (1)

jsavage, 2019-06-27:
I have now found the workaround [in issue 526](https://github.com/google/aiyprojects-raspbian/issues/526#issuecomment-491370978)

eg: the following 4 steps  
```
downloaded protobuf six setuptools
pip3 install ./protobuf-3.8.0-py2.py3-none-any.whl
#now edit out 2 lines as described in #526
sudo nano /var/lib/dpkg/info/aiy-python-wheels.postinst
sudo dpkg --configure --force-overwrite --force-overwrite-dir -a
```

andrewtvuong, 2019-07-04:
@jsavage I've experience the same issues and abandoned this route. Does your work around solve the issue?

dmitriykovalev, 2019-11-14:
That should be now fixed, please check the latest [HACKING.md](https://github.com/google/aiyprojects-raspbian/blob/aiyprojects/HACKING.md), pay attention that apt source URL was changed.

closed 2019-11-14 by dmitriykovalev