google/aiyprojects-raspbian #633: Vision kit not booting after 15mins

dexterac, 2019-06-29:
Bought a new AIY Vision kit 1.1 from Target yesterday. I assembled it as per the instructions and yet the kit wont boot after connecting to 5V 2.4A power supply. 

I am waiting for 15mins and no result.

dexterac, 2019-06-29:
Update :

It took more than 20 mins to get started.

Now when I stop the Joy Detection program and run Face detection i get this error

pi@raspberrypi:~/AIY-projects-python/src/examples/vision $ ./face_camera_trigger.py 

mmal: mmal_vc_component_create: failed to create component 'vc.ril.camera' (1:ENOMEM)
mmal: mmal_component_create_core: could not create component 'vc.ril.camera' (1)
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/picamera/camera.py", line 456, in _init_camera
    self._camera = mo.MMALCamera()
  File "/usr/lib/python3/dist-packages/picamera/mmalobj.py", line 2279, in __init__
    super(MMALCamera, self).__init__()
  File "/usr/lib/python3/dist-packages/picamera/mmalobj.py", line 633, in __init__
    prefix="Failed to create MMAL component %s" % self.component_type)
  File "/usr/lib/python3/dist-packages/picamera/exc.py", line 184, in mmal_check
    raise PiCameraMMALError(status, prefix)
picamera.exc.PiCameraMMALError: Failed to create MMAL component b'vc.ril.camera': Out of memory

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "./face_camera_trigger.py", line 40, in <module>
    main()
  File "./face_camera_trigger.py", line 23, in main
    with PiCamera() as camera:
  File "/usr/lib/python3/dist-packages/picamera/camera.py", line 431, in __init__
    self._init_camera(camera_num, stereo_mode, stereo_decimate)
  File "/usr/lib/python3/dist-packages/picamera/camera.py", line 460, in _init_camera
    "Camera is not enabled. Try running 'sudo raspi-config' "
picamera.exc.PiCameraError: Camera is not enabled. Try running 'sudo raspi-config' and ensure that the camera has been enabled.
pi@raspberrypi:~/AIY-projects-python/src/examples/vision $

andrewtvuong, 2019-07-04:
I find there's inconsistency from time to time with the hardware. Sometimes things work, sometimes they don't. If you have extra parts try swapping out each component. Otherwise take it out of any housing and directly test the parts with tight cable connections.

NagaKanaparthy, 2019-08-10:
I think it is power adapter related. I found that using a 1.5 amp charger would for some reason boot faster and stabler . I think it depends on the power adapter as a lot of them drop below spec for pi voltage stability.

manoj7410, 2019-11-14:
Can you reflash your kit with https://github.com/google/aiyprojects-raspbian/releases/tag/v20191113 and check if the issue is still reproducible ?

dmitriykovalev, 2019-11-18:
Make sure to use good power adapter as specified at https://aiyprojects.withgoogle.com/vision#meet-your-kit--gather-additional-items. Pleas reopen if problem is still reproducible.

closed 2019-11-18 by dmitriykovalev