google/aiyprojects-raspbian #634: Speaker test OK, Microphone test FAILED

baqwas, 2019-06-30:
Hello,

I have the **v1.0** Voice Kit that has served me well since the early days. I've periodically cycled through apt update/upgrade without any issue. Since I was updating my other RPi boards, I decided to do a fresh install with the latest image https://github.com/google/aiyprojects-raspbian/releases/tag/v20181116 Now, the audio test works with the speaker but the microphone test does not echo back my speech "Testing 1 2 3..." as required to conclude the audio test. Needless to add that all this worked when I had used the original image when it was the only one available.

_Is the **v1.0** Voice Kit limited to the first available image?_ https://github.com/google/aiyprojects-raspbian/releases/tag/voicekit

Given that the latest image did not work for my v1.0 installation, I thought it best to seek out advice before going down the route of Raspian Buster followed by the manual installation instructions for support packages and drivers at https://github.com/google/aiyprojects-raspbian/blob/aiyprojects/HACKING.md

I have not reconfigured the latest attempts with Google Cloud API credentials because my newbie interpretation is that this step is only required after I advance past the local tests. I_s my assumption correct?_

Thanks and Kind Regards.

andrewtvuong, 2019-07-04:
I ran into this issue and unfortunately the problem just goes away after several reformating/reimaging with all else being the same.

Don't do manual installs, it's a mess at least for me.

Could you try and see what happens after:
`pkill pulseaudio`

baqwas, 2019-07-13:
Thx  andrewtvuong!

**pkill pulseaudio** did nothing for me under the Buster image. Fell back to the v1 Google image.

The v1 kit is now $2.49 at Micro Center and Target will not lower its price for v1.1 from the astronomical double digit height. Got other fish to fry with FPGA modules and TensorFlow and that's where I will head. I think Android development for Voice would be a better use of my limited time.

baqwas, 2019-07-13:
Google AIY Voice V1 is probably sunset?

closed 2019-07-13 by baqwas