google/aiyprojects-raspbian #635: Buster based image?

workingmanrob, 2019-07-06:
Raspian released buster last week and Debian released it today.

Is an updated image coming soon?

Thanks!

itsalljustaride, 2019-07-08:
Fingers crossed that we get at least one more official build. Either that or they open source this and at least let someone else keep it going. Would hate to have a bunch of hardware that won't run on new builds.

andrewtvuong, 2019-07-09:
@itsalljustaride sorry I don't understand open sourced too well, but what is the line between open sourced and not? I thought the source code to AIY is in this repo. Thanks

itsalljustaride, 2019-07-09:
@andrewtvuong The python code is all OS, and in the repo here, but it's my understanding there are some lower-level hardware things that get pulled down from the deb repos that are not. I could be wrong about that. Hence why so many people can't run the AIY hardware on newer firmware/kernels or use anything but the distributed disk images. Again, I could be wrong about that, but if there is an easy fix for all of these hardware issues, it would be nice to get a walkthrough of how to do it.

workingmanrob, 2019-07-10:
I was about to attempt these instructions on a buster image - https://github.com/google/aiyprojects-raspbian/blob/aiyprojects/HACKING.md - to see if it just works.

Will post back if I make any progress.

itsalljustaride, 2019-07-10:
My RPi 4 just arrived in the mail today, so I thought I'd try the Hacking.md instructions just for funsies, and I'm getting failed installs on all of the deb packages. Doesn't look good.

andrewtvuong, 2019-07-10:
@itsalljustaride 3A+ in the beginning of the year was failing the hacking.md haven’t checked since, but the images do work for it if I remember correctly

jmertic, 2019-07-11:
I upgraded the image to Buster - got an error about 'aiy' not found' but after following the instructions in the HACKING.md it seemed to work fine.

workingmanrob, 2019-07-20:
@jmertic Did you have the assistant working before you updated to buster?

I upgraded a brand fresh 2018-11-16 image to buster, then tried to enable the assistant and it threw an error trying to open chromium to authenticate.

andrewtvuong, 2019-07-20:
@workingmanrob could you authenticate with another device? all mine have been headless

workingmanrob, 2019-07-20:
@andrewtvuong never tried.. mine also runs a magic mirror so typically I control it over VNC.

Since it is wall mounted and a major pain to deal with SD cards I also netboot it using an NFS root that lives on my NAS.. really wishing I had a second voice hat at this point for testing though.

andrewtvuong, 2019-07-20:
@workingmanrob I'm not understanding your setup very well, but you could ssh in, run the any of the assistant programs and when it asks you to authenticate with a url, copypaste it onto your computer's browser and authenticate from there.

workingmanrob, 2019-07-20:
@andrewtvuong that URL that I normally see before the browser is automatically opened is never displayed.

It just threw an error.. like got 3 arguments when 4 were expected or something like that.  Might have been got 4 expecting 3 but I don't remember offhand.

The NFS setup is really irrelevant.  It's just awesome.  SD card corruption never happens cause there is no SD card.

andrewtvuong, 2019-07-20:
@workingmanrob wow that's really cool, I'm always trying to prioritize and maximize reliability. Got a link about the NFS setup so I can read up on how?

workingmanrob, 2019-07-21:
@andrewtvuong sure thing.  The rpi folks have a page about it..

https://www.raspberrypi.org/documentation/hardware/raspberrypi/bootmodes/net_tutorial.md

the important bits are to make sure your pi has netboot enabled.

`$ vcgencmd otp_dump | grep 17:`
`17:3020000a`

then modify /boot/cmdline.txt (make a backup!)
`$ sudo cp /boot/cmdline.txt /boot/cmdline.txt.org`
`$ sudo vi /boot/cmdline.txt`
Change the root=PARTUUID=blahblah rootfstype=ext4 stuff to
`root=/dev/nfs nfsroot=NFS_SERVER_IP:/path/to/nfsroot,rsize=16384,wsize=16384,tcp,vers=3 rw ip=dhcp rootfstype=nfs`

Shut down your pi.  Copy the entire sdcard contents to NAS shared folder and fixup /etc/fstab in nfsroot.
`proc            /proc           proc    defaults          0       0`
`/dev/mmcblk0p1  /boot           vfat    defaults          0       2`

You can also clean out /dev/* /proc/* /sys/* /tmp/* plus rmdir the /debootstrap and /lost+found folders from NAS nfsroot.

Put your sdcard back in your Pi for now and when you power it up it should be using nfsroot from your NAS.

With it like this you're not really netbooting yet.  You still have to setup tftp to serve /boot but this makes it really easy to flip beween the card and the nfsroot.
`$ sudo cp /boot/cmdline.txt /boot/cmdline.txt.nfs`
`$ sudo cp /boot/cmdline.txt.org /boot/cmdline.txt`
`$ sudo reboot`
And you're booting from the sdcard again.

I never did setup tftp cause I found that part not 100% reliable but now I am regretting that decision cause I can't easily flip from stretch to buster and back by just changing the tftp folder served.

Really I'm hoping the new RPi4 model (which I personally haven't got ahold of yet) plus the release of buster will mean we see a new aiy release and I can do a full rebuild from scratch.

jmertic, 2019-07-22:
> @jmertic Did you have the assistant working before you updated to buster?
> 
> I upgraded a brand fresh 2018-11-16 image to buster, then tried to enable the assistant and it threw an error trying to open chromium to authenticate.

Yes - had it working fine before and just needed to follow the HACKING instructions to get the module back configured.

tarzanek, 2019-07-22:
did someone get the dkms modules patched for new kernel? https://github.com/google/aiyprojects-raspbian/issues/627

ginkage, 2019-07-23:
@tarzanek I got it working for me on 4.19.57, see the last comment on that issue.

workingmanrob, 2019-07-25:
So I attempted to update a working voice kit to buster.  The assistant demo script just throws a segmentation fault.

@dmitriykovalev any hope of a buster image update?  Even buster specific hacking instructions would do me just fine.

olafthiele, 2019-08-03:
New image would be greatly appreciated :-)

workingmanrob, 2019-08-15:
Hey google - Where is that buster image?

No response from my voice kit... or google.  Sigh.

workingmanrob, 2019-08-23:
I have it working.

Started with the full buster image from 2019-06-20 and am just testing again using the latest image from 2019-07-10.  If all goes well I'll share my step by step instructions.

Thanks for all the support Google. :-1:

workingmanrob, 2019-08-23:
Actually working HACKING.md instructions!

After you have buster installed and updated follow the info below.

echo "deb https://dl.google.com/aiyprojects/deb stable main" | sudo tee /etc/apt/sources.list.d/aiyprojects.list
wget -q -O - https://dl.google.com/linux/linux_signing_key.pub | sudo apt-key add -

sudo apt-get update
sudo apt-get upgrade

reboot

sudo apt-get install -y aiy-voicebonnet-soundcard-dkms aiy-dkms

Apply the fix - https://github.com/google/aiyprojects-raspbian/issues/627#issuecomment-514403972

sudo cp /usr/src/aiy-1.1/aiy/iio/adc/aiy-adc.c /usr/src/aiy-1.1/aiy/iio/adc/aiy-adc.c.org
sudo vi +138 /usr/src/aiy-1.1/aiy/iio/adc/aiy-adc.c # delete that line

sudo cp /usr/src/aiy-1.1/aiy/mfd/aiy-io-i2c.c /usr/src/aiy-1.1/aiy/mfd/aiy-io-i2c.c.org
sudo vi /usr/src/aiy-1.1/aiy/mfd/aiy-io-i2c.c

Replace two occurrences (in aiy_io_reset and in aiy_io_update_firmware) of
i2c_lock_adapter(i2c->adapter);
with
i2c_lock_bus(i2c->adapter, I2C_LOCK_ROOT_ADAPTER);

and replace two occurrences of
i2c_unlock_adapter(i2c->adapter);
with
i2c_unlock_bus(i2c->adapter, I2C_LOCK_ROOT_ADAPTER);

sudo dkms install aiy/1.1

- the aiy-voicebonnet-soundcard-dkms will still fail.  But who cares?!  Apparently it does nothing I care about... you just need the following in /boot/config.txt

dtoverlay=googlevoicehat-soundcard

reboot

sudo apt-get install -y pulseaudio pavucontrol
sudo mkdir /etc/pulse/daemon.conf.d/
echo "default-sample-rate = 48000" | sudo tee /etc/pulse/daemon.conf.d/aiy.conf

sudo apt-get install -y aiy-python-wheels

Apply the fix - https://github.com/google/aiyprojects-raspbian/issues/526#issuecomment-491370978

done as the pi user
cd Downloads
pip3 download protobuf
pip3 install ./protobuf-3.9.1-py2.py3-none-any.whl
sudo vi /var/lib/dpkg/info/aiy-python-wheels.postinst

Comment out the lines containing

`# pip3 install --no-deps --no-cache-dir --disable-pip-version-check \`
`#             /opt/aiy/python-wheels/protobuf-3.6.1-cp35-cp35m-linux_armv6l.whl`

sudo dpkg --configure --force-overwrite --force-overwrite-dir -a

reboot

sudo apt-get install -y leds-ktd202x-dkms
sudo apt-get install -y pwm-soft-dkms

reboot

git clone https://github.com/google/aiyprojects-raspbian.git AIY-projects-python

To match old SD card layout...
sudo mv AIY-projects-python /opt/aiy/projects-python
ln -s /opt/aiy/projects-python AIY-projects-python
ln -s /home/pi/AIY-projects-python AIY-voice-kit-python

sudo pip3 install -e AIY-projects-python/src

reboot

Put your assistant.json file in /home/pi

sudo cp /opt/aiy/projects-python/src/aiy/assistant/auth_helpers.py /opt/aiy/projects-python/src/aiy/assistant/auth_helpers.py.org
sudo vi +75 /opt/aiy/projects-python/src/aiy/assistant/auth_helpers.py ### remove the , -1 at the end of the line

src/examples/voice/assistant_grpc_demo.py ### this worked and let me authenticate with google.

src/examples/voice/assistant_library_with_button_demo.py ### this failed with module error
ModuleNotFoundError: No module named 'google.assistant.library'

pi@mi-berry:~ $ pip3 install google-assistant-library

This failed and when I tried with sudo it still didn't work but then I did this:

pi@mi-berry:~ $ pip3 install google-assistant-library==1.0.0
pi@mi-berry:~ $ pip3 install google-assistant-library

src/examples/voice/assistant_library_demo.py ### WORKING!!!

No idea why it works after doing that but at this point I really don't fucking care.
**Suck a big bag of dicks google.**

mickey9801, 2019-10-17:
Thanks @workingmanrob . One note: I need to install gprcio verions 1.23.0 otherwise the assistant_grpc_demo.py demo will get error.

```
sudo pip install grpcio==1.23.0
```

Paprika-Asteria, 2019-11-13:
if i can't load google.com,how i can install this aiy?

dmitriykovalev, 2019-11-14:
There is a Buster-based image now, please check [Releases](https://github.com/google/aiyprojects-raspbian/releases) section.

surak, 2019-11-17:
> There is a Buster-based image now, please check [Releases](https://github.com/google/aiyprojects-raspbian/releases) section.

Thanks Dmitriy. Would you care to comment on https://github.com/google/aiyprojects-raspbian/issues/630 ? What's the official word on the continuity of this project?

peopledrivemecrazy, 2019-12-11:
@dmitriykovalev I am using the latest image but @workingmanrob https://github.com/google/aiyprojects-raspbian/issues/635#issuecomment-524366391 works. 

Why not release a better build with fixes.

workingmanrob, 2020-01-07:
Hey guys,

To close this completely I am now using the provided AIY image from 2019-11-13 and it still ships with some broken bits.

`cp AIY-projects-python/src/aiy/assistant/auth_helpers.py AIY-projects-python/src/aiy/assistant/auth_helpers.py.org`
`vi +75 AIY-projects-python/src/aiy/assistant/auth_helpers.py`
remove the `, -1` at the end of the line

With your assistant.json file in place you should be able to run any of the demo's now and get authorized.  Then it will seg fault.  Fix is to run this as the pi user.

`pip3 install google-assistant-library`
`pip3 install google-assistant-library==1.0.0`
`pip3 install google-assistant-library`

Now things will work as they should.

That should really have been included in the "image".

HTH

closed 2020-01-07 by workingmanrob

Paprika-Asteria, 2020-01-08:
thanks,i have combine usb coral with rpi4,its so good.------------------ 原始邮件 ------------------

 
 
 
 
 
 
 
 
&mdash;