google/aiyprojects-raspbian #640: Raspbain Installation Error

Piflyer, 2019-08-13:
I tried following HACKING.md, and when I downed aiy-dkms, it came out an error like this: `Setting up aiy-dkms (1.1-2) ...
Removing old aiy-1.1 DKMS files...

------------------------------
Deleting module version: 1.1
completely from the DKMS tree.
------------------------------
Done.
Loading new aiy-1.1 DKMS files...
It is likely that 4.19.58-v7l+ belongs to a chroot's host
Building for 4.19.58+, 4.19.58-v7+ and 4.19.58-v7l+
Building initial module for 4.19.58+
Error! Bad return status for module build on kernel: 4.19.58+ (armv7l)
Consult /var/lib/dkms/aiy/1.1/build/make.log for more information.
dpkg: error processing package aiy-dkms (--configure):
 installed aiy-dkms package post-installation script subprocess returned error exit status 10
Errors were encountered while processing:
 aiy-dkms
` I'm using a new Buster image for my Raspberry Pi 4. Can someone please help? Thanks.

KPTKleist, 2019-08-31:
I also encountered the same issue #643, it may have something to do with the incompatibility with the latest kernel.

dmitriykovalev, 2019-11-14:
Should be fixed in the latest SD card image, check [Releases](https://github.com/google/aiyprojects-raspbian/releases) section and updated [HACKING.md](https://github.com/google/aiyprojects-raspbian/blob/aiyprojects/HACKING.md).

closed 2019-11-14 by dmitriykovalev