google/aiyprojects-raspbian #641: AIY vision kit error can't run the demo

DaXiongAhey, 2019-08-18:
I flashed the latest image to the SD card. And I didn't `sudo apt-get update` or `sudo apt-get upgrade`, because some issues have shown that update or upgrade would cause any other problems.
But even I checked the connecters "thousands times", it always had failure to run the program correctly. So I made a video for this mistake about what it happened after I run the demo:
https://youtu.be/zpCFq9uftHY

The log is here:

> -- Logs begin at Sun 2019-08-18 10:20:56 UTC. --
Aug 18 10:59:37 raspberrypi python3[13978]:     if cb(*exc_details):
Aug 18 10:59:37 raspberrypi python3[13978]:   File "/usr/lib/python3.5/contextlib.py", line 261, in _exit_wrapper
Aug 18 10:59:37 raspberrypi python3[13978]:     return cm_exit(cm, *exc_details)
Aug 18 10:59:37 raspberrypi python3[13978]:   File "/usr/lib/python3/dist-packages/picamera/camera.py", line 758, in __exit__
Aug 18 10:59:37 raspberrypi python3[13978]:     self.close()
Aug 18 10:59:37 raspberrypi python3[13978]:   File "/usr/lib/python3/dist-packages/picamera/camera.py", line 752, in close
Aug 18 10:59:37 raspberrypi python3[13978]:     raise exc
Aug 18 10:59:37 raspberrypi python3[13978]:   File "/usr/lib/python3/dist-packages/picamera/camera.py", line 1995, in _control_callback
Aug 18 10:59:37 raspberrypi python3[13978]:     "No data recevied from sensor. Check all connections, "
Aug 18 10:59:37 raspberrypi python3[13978]: picamera.exc.PiCameraRuntimeError: No data recevied from sensor. Check all connections, including the SUNNY chip on the camera board

I have no idea what  I can do for troubleshooting. Please help me. Thanks!

manoj7410, 2019-09-03:
@DaXiongAhey Can you try to connect the camera directly to RPI and run below command :

raspistill -o file.jpg

Please share what happens.

dmitriykovalev, 2019-11-14:
Please try to use new SD card image [aiyprojects-2019-11-13.img.xz](https://github.com/google/aiyprojects-raspbian/releases/download/v20191113/aiyprojects-2019-11-13.img.xz) at [Releases](https://github.com/google/aiyprojects-raspbian/releases) section.

closed 2019-11-14 by dmitriykovalev