google/aiyprojects-raspbian #643: aiy-dkms Installation Error Related to Kernel

KPTKleist, 2019-08-31:
I tried to set up the AIY voice project on my Raspberry Pi 4B, since the AIY img is outdated for RPi 4B, I had to follow the instruction in Hacking.md to set it up on a existing Raspbian Buster that was freshly installed. Then I ran into some troubles installing aiy-dkms, it seemed like aiy-dkms wasn't able to build on kernel 4.19.66+ (armv7l) which is the latest version as I'm writing this issue. A similar issue also happened during the installation of aiy-voicebonnet-soundcard-dkms.

`root@raspberrypi:/home/pi# apt-get install aiy-dkms`
`Reading package lists... Done`
`Building dependency tree`
`Reading state information... Done`
`The following package was automatically installed and is no longer required:`
  `rpi.gpio-common`
`Use 'sudo apt autoremove' to remove it.`
`The following NEW packages will be installed:`
 ` aiy-dkms`
`0 upgraded, 1 newly installed, 0 to remove and 3 not upgraded.`
`Need to get 0 B/12.0 kB of archives.`
`After this operation, 80.9 kB of additional disk space will be used.`
`Selecting previously unselected package aiy-dkms.`
`(Reading database ... 203193 files and directories currently installed.)`
`Preparing to unpack .../aiy-dkms_1.1-2_all.deb ...`
`Unpacking aiy-dkms (1.1-2) ...`
`Setting up aiy-dkms (1.1-2) ...`
`Loading new aiy-1.1 DKMS files...`
`It is likely that 4.19.66-v7l+ belongs to a chroot's host`
`Building for 4.19.66+, 4.19.66-v7+ and 4.19.66-v7l+`
`Building initial module for 4.19.66+`
`Error! Bad return status for module build on kernel: 4.19.66+ (armv7l)`
`Consult /var/lib/dkms/aiy/1.1/build/make.log for more information.`
`dpkg: error processing package aiy-dkms (--configure):`
` installed aiy-dkms package post-installation script subprocess returned error exit status 10`
`Errors were encountered while processing:`
` aiy-dkms`
`E: Sub-process /usr/bin/dpkg returned an error code (1)`

zikotrance, 2019-09-27:
Same thing here. Do you had any answer somewhere else to help resolving the issue?
Cheers

tarzanek, 2019-10-01:
https://github.com/google/aiyprojects-raspbian/issues/627#issuecomment-514403972

dmitriykovalev, 2019-11-14:
This should be fixed in the latest [aiyprojects-2019-11-13.img.xz](https://github.com/google/aiyprojects-raspbian/releases/download/v20191113/aiyprojects-2019-11-13.img.xz). Debian packages are now hosted at https://packages.cloud.google.com/, check updated [HACKING.md](https://github.com/google/aiyprojects-raspbian/blob/aiyprojects/HACKING.md).

closed 2019-11-14 by dmitriykovalev