google/aiyprojects-raspbian #645: Raspberry Pi 4 and AIY voice kit V1

Selinoid, 2019-09-15:
I build robots and use AIY voice hat with Rpi3b+. I just bought a Rpi4 I wonder if the AIY voice kit is compatible with the RPi4. Thanks in advance.

zikotrance, 2019-09-27:
Same thing here. Do you had any answer somewhere else to help resolving the issue?
Cheers

tarzanek, 2019-10-01:
https://github.com/google/aiyprojects-raspbian/issues/627#issuecomment-514403972

tupelotech, 2019-11-02:
v20181116 and

v20170911 images certainly do not boot on Pi4

tarzanek, 2019-11-08:
> 
> 
> v20181116 and
> 
> v20170911 images certainly do not boot on Pi4

you need latest raspbian for rpi4, so those images need to be updated first and patched as per https://github.com/google/aiyprojects-raspbian/issues/627#issuecomment-514403972 

hth
L

Paprika-Asteria, 2019-11-15:
i have tried install aiy-vision in RPi4,but have some problems,i didn't install 'protobuf-3.6.1-cp35-cp35m-linux_armv6l.whl' beacuse i need armv7l,and i have protobuf 3.5 ,i get into trouble,i need help plz

tarzanek, 2019-11-15:
@Paprika-Asteria check https://github.com/google/aiyprojects-raspbian/releases/tag/v20191113
it will work

klomfar, 2019-11-17:
hi, i get this error :
pi@raspberrypi:~ $ sudo dtoverlay googlevoicehat-soundcard
* Failed to apply overlay '0_googlevoicehat-soundcard' (kernel)
i have RPi 4 with Voice Kit V1, please help :(

Paprika-Asteria, 2019-11-18:
> 
> 
> @Paprika-Asteria check https://github.com/google/aiyprojects-raspbian/releases/tag/v20191113
> it will work

i have install this image,but my when i run this examples ,my rpi4 don't work,
"FileNotFoundError: [Errno 2] No such file or directory: '/dev/vision_spicomm'"
i can't find this file,i need help.thanks

dmitriykovalev, 2019-11-18:
@klomfar My Voice HAT works with Raspberry Pi 4 without any issues. Can you please try to run and copy output from
```
lsmod | grep snd_soc_googlevoicehat_codec
```
before and after executing overlay load command (`sudo dtoverlay googlevoicehat-soundcard`).

In addition, can you attach `dmesg` command output when it fails?

ziligy, 2019-11-20:
Just to confirm release 2019-11-13 works on my Pi 4 with Voice hat v1  :)

klomfar, 2019-11-21:
it works now,thanks for help @dmitriykovalev

dmitriykovalev, 2019-11-21:
Great! Closing this issue then.

closed 2019-11-21 by dmitriykovalev

Paprika-Asteria, 2019-11-22:
but i havenot this kit,i have only rpi4.it donot work.whether you install this image on your rpi4?i have a trouble.━(◯Δ◯∥)------------------ 原始邮件 ------------------

 
&mdash;