google/aiyprojects-raspbian #647: RaspberryPi 4 AIY Voice Hat not working

zikotrance, 2019-09-27:
Hi guys, I'm new here and been struggled with making the mic and speaker working.

Seems that Google is no longer supporting Phyton and push us to use Google services instead. In other words I'm not able to install manually the AIY-Project to add it to my smart mirror. Do you guys have any work around with guidance because I'm new in the coding world

Thanks

everybodyfloats, 2019-09-29:
Im having the same issue. No lights on the button and no audio after adding 'dtoverlay=googlevoicehat-soundcard" to  /boot/config.txt

tarzanek, 2019-10-01:
https://github.com/google/aiyprojects-raspbian/issues/627#issuecomment-514403972

tarzanek, 2019-11-15:
https://github.com/google/aiyprojects-raspbian/releases/tag/v20191113 looks like it could work

Paprika-Asteria, 2019-11-18:
i have install this image v20191113 on my rpi4,but i can't use this aiy-vision. File "/home/pi/AIY-projects-python/src/examples/vision/face_detection_raspivid.py", line 62, in <module>
    main()
  File "/home/pi/AIY-projects-python/src/examples/vision/face_detection_raspivid.py", line 55, in main
    CameraInference(face_detection.model()) as inference:
  File "/home/pi/AIY-projects-python/src/aiy/vision/inference.py", line 101, in __init__
    self._engine = self._stack.enter_context(InferenceEngine())
  File "/home/pi/AIY-projects-python/src/aiy/vision/inference.py", line 275, in __init__
    self._transport = make_transport()
  File "/home/pi/AIY-projects-python/src/aiy/vision/_transport.py", line 86, in make_transport
    return _SpiTransport()
  File "/home/pi/AIY-projects-python/src/aiy/vision/_transport.py", line 28, in __init__
    self._spicomm = _spicomm.Spicomm()
  File "/home/pi/AIY-projects-python/src/aiy/vision/_spicomm.py", line 317, in __init__
    super().__init__()
  File "/home/pi/AIY-projects-python/src/aiy/vision/_spicomm.py", line 219, in __init__
    self._dev = os.open(SPICOMM_DEV, os.O_RDWR)
FileNotFoundError: [Errno 2] No such file or directory: '/dev/vision_spicomm'

dmitriykovalev, 2019-11-18:
@Paprika-Asteria Vision Bonnet is not compatible with Raspberry Pi 4, so it's not going to work. The original issue was about Voice HAT.

AsidRum, 2019-11-30:
doesn't work, i use aiyprojects-2019-11-13.img.xz
image, and i get this
```
pi@raspberrypi:~ $ sudo dtoverlay googlevoicehat-soundcard
* Failed to apply overlay '0_googlevoicehat-soundcard' (kernel)
```
maybe this info will help
```
pi@raspberrypi:~ $ lsmod | grep snd_soc_googlevoicehat_codec
snd_soc_googlevoicehat_codec    16384  1
snd_soc_core          192512  4 snd_soc_googlevoicehat_codec,vc4,snd_soc_bcm2835_i2s,snd_soc_rpi_simple_soundcard
```

```
pi@raspberrypi:~ $ dmesg
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.19.75-v7l+ (dom@buildbot) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611)) #1270 SMP Tue Sep 24 18:51:41 BST 2019
[    0.000000] CPU: ARMv7 Processor [410fd083] revision 3 (ARMv7), cr=30c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, PIPT instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 4 Model B Rev 1.1
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] cma: Reserved 256 MiB at 0x000000001ec00000
[    0.000000] On node 0 totalpages: 229376
[    0.000000]   DMA zone: 1728 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 196608 pages, LIFO batch:63
[    0.000000]   HighMem zone: 32768 pages, LIFO batch:7
[    0.000000] random: get_random_bytes called from start_kernel+0xc0/0x4e8 with crng_init=0
[    0.000000] percpu: Embedded 17 pages/cpu s39488 r8192 d21952 u69632
[    0.000000] pcpu-alloc: s39488 r8192 d21952 u69632 alloc=17*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 227648
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 cma=64M cma=256M  smsc95xx.macaddr=DC:A6:32:0E:2A:04 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyS0,115200 console=tty1 root=PARTUUID=d9b3f436-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Memory: 629692K/917504K available (8192K kernel code, 661K rwdata, 2352K rodata, 2048K init, 850K bss, 25668K reserved, 262144K cma-reserved, 131072K highmem)
[    0.000000] Virtual kernel memory layout:
                   vector  : 0xffff0000 - 0xffff1000   (   4 kB)
                   fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
                   vmalloc : 0xf0800000 - 0xff800000   ( 240 MB)
                   lowmem  : 0xc0000000 - 0xf0000000   ( 768 MB)
                   pkmap   : 0xbfe00000 - 0xc0000000   (   2 MB)
                   modules : 0xbf000000 - 0xbfe00000   (  14 MB)
                     .text : 0x(ptrval) - 0x(ptrval)   (10208 kB)
                     .init : 0x(ptrval) - 0x(ptrval)   (2048 kB)
                     .data : 0x(ptrval) - 0x(ptrval)   ( 662 kB)
                      .bss : 0x(ptrval) - 0x(ptrval)   ( 851 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 28124 entries in 83 pages
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] arch_timer: cp15 timer(s) running at 54.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xc743ce346, max_idle_ns: 440795203123 ns
[    0.000006] sched_clock: 56 bits at 54MHz, resolution 18ns, wraps every 4398046511102ns
[    0.000018] Switching to timer-based delay loop, resolution 18ns
[    0.000242] Console: colour dummy device 80x30
[    0.000266] console [tty1] enabled
[    0.000306] Calibrating delay loop (skipped), value calculated using timer frequency.. 108.00 BogoMIPS (lpj=540000)
[    0.000328] pid_max: default: 32768 minimum: 301
[    0.000587] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.000605] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.001345] CPU: Testing write buffer coherency: ok
[    0.001752] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.002392] Setting up static identity map for 0x200000 - 0x20003c
[    0.002548] rcu: Hierarchical SRCU implementation.
[    0.003405] smp: Bringing up secondary CPUs ...
[    0.004297] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[    0.005321] CPU2: thread -1, cpu 2, socket 0, mpidr 80000002
[... 289 lines cut ...]
[   15.423154] Bluetooth: Core ver 2.22
[   15.423199] NET: Registered protocol family 31
[   15.423203] Bluetooth: HCI device and connection manager initialized
[   15.423216] Bluetooth: HCI socket layer initialized
[   15.423222] Bluetooth: L2CAP socket layer initialized
[   15.423242] Bluetooth: SCO socket layer initialized
[   15.432538] Bluetooth: HCI UART driver ver 2.3
[   15.432544] Bluetooth: HCI UART protocol H4 registered
[   15.432578] Bluetooth: HCI UART protocol Three-wire (H5) registered
[   15.432683] Bluetooth: HCI UART protocol Broadcom registered
[   15.576582] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   15.576588] Bluetooth: BNEP filters: protocol multicast
[   15.576599] Bluetooth: BNEP socket layer initialized
[   15.595334] Bluetooth: RFCOMM TTY layer initialized
[   15.595356] Bluetooth: RFCOMM socket layer initialized
[... cut ...]
[  154.192934] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[  219.984816] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /soc/i2s@7e203000/status
[  268.346394] voicehat-codec voicehat-codec: Enabling audio amp...
[  268.354840] voicehat-codec voicehat-codec: Enabling audio amp...
[  297.563410] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /soc/i2s@7e203000/status
[ 2580.764093] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /soc/i2s@7e203000/status

```

dmitriykovalev, 2019-12-03:
@AsidRum Which board do you have: Voice Bonnet (Pi Zero like size) or Voice HAT (Pi3 like size)?

AsidRum, 2019-12-03:
@dmitriykovalev Voice HAT

nunziofiore, 2020-05-25:
any news guys? we had same problem

AsidRum, 2020-05-25:
Try to manually install all components on fresh raspbian. For me it's start working. I have only one issue microphone level is low

KonsultInternetConsulting, 2020-05-27:
AsidRum Is there somewhere to show us that?