google/aiyprojects-raspbian #648: No drivers for ALSA sound card

elhanafi5480, 2019-10-18:
When I run the lmage without the Google hat , and run  ' aplay -l ' ALSA sound card is not exist ,so I can't play sound through HDMI or 3.5mm input sound

manoj7410, 2019-10-31:
Which version of RPI do you have ?
Have you reflashed your AIY board with the image provided at https://github.com/google/aiyprojects-raspbian/releases/tag/v20181116 ?

elhanafi5480, 2019-11-12:
yes,
when running 'aplay -l' through the terminal , i see this result that ensure ALSA sound card drivers not exist.
**** List of PLAYBACK Hardware Devices ****

card 1: Device [USB PnP Sound Device], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

i use external sound card

dmitriykovalev, 2019-11-14:
Please consult the documentation for external sound card. You can uncomment `dtparam=audio=on` in `/boot/config.txt` to enable Pi builtin sound output.

closed 2019-11-14 by dmitriykovalev