google/aiyprojects-raspbian #649: Error voice/assistant_grpc_demo.py

afrdt, 2019-10-23:
DEBUG:requests_oauthlib.oauth2_session:Requesting url https://oauth2.googleapis.com/token using method POST.
DEBUG:requests_oauthlib.oauth2_session:Supplying headers {'Content-Type': 'application/x-www-form-urlencoded;charset=UTF-8', 'Accept': 'application/json'} and data {'grant_type': 'authorization_code', 'redirect_uri': 'urn:ietf:wg:oauth:2.0:oob', 'client_secret': 'xk663HAK4ooulPuONgxIz8ly', 'code': '4/sQEvtQfls-if6UygeD3sdioGnT67uZw8EZYomEovr5Gz4GSB1FwbJ3A', 'client_id': '84329723989-cunk66tj22q9ijq3figj0gl2bpq47qur.apps.googleusercontent.com'}
DEBUG:requests_oauthlib.oauth2_session:Passing through key word arguments {'timeout': None, 'json': None, 'verify': True, 'auth': None, 'proxies': None}.
DEBUG:urllib3.connectionpool:Starting new HTTPS connection (1): oauth2.googleapis.com:443
Traceback (most recent call last):
  File "/usr/local/lib/python3.5/dist-packages/urllib3/contrib/pyopenssl.py", line 453, in wrap_socket
    cnx.do_handshake()
  File "/usr/lib/python3/dist-packages/OpenSSL/SSL.py", line 1426, in do_handshake
    self._raise_ssl_error(self._ssl, result)
  File "/usr/lib/python3/dist-packages/OpenSSL/SSL.py", line 1174, in _raise_ssl_error
    _raise_current_error()
  File "/usr/lib/python3/dist-packages/OpenSSL/_util.py", line 48, in exception_from_error_queue
    raise exception_type(errors)
OpenSSL.SSL.Error: [('SSL routines', 'tls_process_server_certificate', 'certificate verify failed')]

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.5/dist-packages/urllib3/connectionpool.py", line 600, in urlopen
    chunked=chunked)
  File "/usr/local/lib/python3.5/dist-packages/urllib3/connectionpool.py", line 343, in _make_request
    self._validate_conn(conn)
  File "/usr/local/lib/python3.5/dist-packages/urllib3/connectionpool.py", line 839, in _validate_conn
    conn.connect()
  File "/usr/local/lib/python3.5/dist-packages/urllib3/connection.py", line 344, in connect
    ssl_context=context)
  File "/usr/local/lib/python3.5/dist-packages/urllib3/util/ssl_.py", line 342, in ssl_wrap_socket
    return context.wrap_socket(sock, server_hostname=server_hostname)
  File "/usr/local/lib/python3.5/dist-packages/urllib3/contrib/pyopenssl.py", line 459, in wrap_socket    raise ssl.SSLError('bad handshake: %r' % e)
ssl.SSLError: ("bad handshake: Error([('SSL routines', 'tls_process_server_certificate', 'certificate verify failed')],)",)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.5/dist-packages/requests/adapters.py", line 449, in send
    timeout=timeout
  File "/usr/local/lib/python3.5/dist-packages/urllib3/connectionpool.py", line 638, in urlopen
    _stacktrace=sys.exc_info()[2])
  File "/usr/local/lib/python3.5/dist-packages/urllib3/util/retry.py", line 398, in increment
    raise MaxRetryError(_pool, url, error or ResponseError(cause))
urllib3.exceptions.MaxRetryError: HTTPSConnectionPool(host='oauth2.googleapis.com', port=443): Max retries exceeded with url: /token (Caused by SSLError(SSLError("bad handshake: Error([('SSL routines', 'tls_process_server_certificate', 'certificate verify failed')],)",),))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "./assistant_grpc_demo.py", line 57, in <module>
    main()
  File "./assistant_grpc_demo.py", line 49, in main
    language_code=args.language)
  File "/opt/aiy/projects-python/src/aiy/assistant/grpc.py", line 205, in __init__
    super().__init__(language_code, volume_percentage)
  File "/opt/aiy/projects-python/src/aiy/assistant/grpc.py", line 64, in __init__
    credentials = auth_helpers.get_assistant_credentials()
  File "/opt/aiy/projects-python/src/aiy/assistant/auth_helpers.py", line 132, in get_assistant_credentials
    return _try_to_get_credentials(credentials_file)
  File "/opt/aiy/projects-python/src/aiy/assistant/auth_helpers.py", line 123, in _try_to_get_credentials
    credentials = _credentials_flow_interactive(client_secrets)
  File "/opt/aiy/projects-python/src/aiy/assistant/auth_helpers.py", line 78, in _credentials_flow_interactive
    credentials = flow.run_console()
  File "/usr/local/lib/python3.5/dist-packages/google_auth_oauthlib/flow.py", line 364, in run_console
    self.fetch_token(code=code)
File "/usr/local/lib/python3.5/dist-packages/google_auth_oauthlib/flow.py", line 241, in fetch_token
    self.client_config['token_uri'], **kwargs)
  File "/usr/lib/python3/dist-packages/requests_oauthlib/oauth2_session.py", line 221, in fetch_token
    verify=verify, proxies=proxies)
  File "/usr/local/lib/python3.5/dist-packages/requests/sessions.py", line 572, in post
    return self.request('POST', url, data=data, json=json, **kwargs)
  File "/usr/lib/python3/dist-packages/requests_oauthlib/oauth2_session.py", line 358, in request
    headers=headers, data=data, **kwargs)
  File "/usr/local/lib/python3.5/dist-packages/requests/sessions.py", line 524, in request
    resp = self.send(prep, **send_kwargs)
  File "/usr/local/lib/python3.5/dist-packages/requests/sessions.py", line 637, in send
    r = adapter.send(request, **kwargs)
  File "/usr/local/lib/python3.5/dist-packages/requests/adapters.py", line 514, in send
    raise SSLError(e, request=request)
requests.exceptions.SSLError: HTTPSConnectionPool(host='oauth2.googleapis.com', port=443): Max retries exceeded with url: /token (Caused by SSLError(SSLError("bad handshake: Error([('SSL routines', 'tls_process_server_certificate', 'certificate verify failed')],)",),))

manoj7410, 2019-11-14:
Please take care of the below points:

- Make sure that your kit is connected to an unrestricted network.
- Make sure that you have allowed the grpc demo to access your Google account by pasting the URL (Given by grpc demo) into a browser .
- Paste the authorisation code on the terminal. 
- And please use the latest build that can be found at https://github.com/google/aiyprojects-raspbian/releases/tag/v20191113

Yukiyukiyeah, 2020-03-11:
I also met this error. Did u fix it?

Yukiyukiyeah, 2020-03-11:
> Please take care of the below points:
>
> * Make sure that your kit is connected to an unrestricted network.
> * Make sure that you have allowed the grpc demo to access your Google account by pasting the [...]

I wonder if it is because I use my mobile phone's hotspot for the devices to access to Internet

afrdt, 2020-03-11:
I did fix it. I was using my university's open wifi network and I had to register the device's MAC address in order for it to work.

closed 2020-03-11 by afrdt