google/aiyprojects-raspbian #650: assistant library replacement? (hotword detection is dead with buster baby?)

tarzanek, 2019-11-15:
With recent https://github.com/google/aiyprojects-raspbian/releases/tag/v20191113 image gRPC demo works for me, but ...

all the library demos do Segmentation fault
```
pi@raspberrypi:~ $ ./AIY-projects-python/src/examples/voice/assistant_library_with_button_demo.py
Segmentation fault
```

https://github.com/googlesamples/assistant-sdk-python/issues/356
tries to solve the issue and 
https://developers.google.com/assistant/sdk/reference/library/python/
now has a deprecation notice

so ... is there some reasonable alternative?
or will gRPC library offer the same functionality as assistant one?
@dmitriykovalev  any clues?

thanks for getting the (hotword) stuff work again :-)

tarzanek, 2020-01-08:
I guess a workaround is in 
https://github.com/google/aiyprojects-raspbian/issues/657
but then a fix would be good too, or is https://snowboy.kitt.ai/ the only alternative now?

mdbrewer, 2020-03-13:
Check out my writeup on how to get it working and update stretch. https://medium.com/@michaeldbrewer/aiy-voice-deprecation-workaround-bd07559403df

tarzanek, 2020-03-16:
> 
> 
> Check out my writeup on how to get it working and update stretch. https://medium.com/@michaeldbrewer/aiy-voice-deprecation-workaround-bd07559403df

a bit like scratching around your head and you still have obsolete google library, but I guess it is a workaround (not the one I'd choose, I'd prefer Google to go and fix their libs)