google/aiyprojects-raspbian #651: IsADirectoryError: [Errno 21] Is a directory: '/home/pi/googleassistant/credentials.json'

Trancez15, 2019-11-23:
this is the command in this -> (env) pi@raspberrypi:~ $ 
google-oauthlib-tool --client-secrets ~/googleassistant/credentials.json \
--scope https://www.googleapis.com/auth/assistant-sdk-prototype \
--scope https://www.googleapis.com/auth/gcm \
--save --headless

this is the error:
Traceback (most recent call last):
  File "/home/pi/env/bin/google-oauthlib-tool", line 8, in <module>
    sys.exit(main())
  File "/home/pi/env/lib/python3.5/site-packages/click/core.py", line 722, in __call__
    return self.main(*args, **kwargs)
  File "/home/pi/env/lib/python3.5/site-packages/click/core.py", line 697, in main
    rv = self.invoke(ctx)
  File "/home/pi/env/lib/python3.5/site-packages/click/core.py", line 895, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/pi/env/lib/python3.5/site-packages/click/core.py", line 535, in invoke
    return callback(*args, **kwargs)
  File "/home/pi/env/lib/python3.5/site-packages/google_auth_oauthlib/tool/__main__.py", line 100, in main
    scopes=scope
  File "/home/pi/env/lib/python3.5/site-packages/google_auth_oauthlib/flow.py", line 196, in from_client_secrets_file
    with open(client_secrets_file, 'r') as json_file:
IsADirectoryError: [Errno 21] Is a directory: '/home/pi/googleassistant/credentials.json'

How can i fix it?

manoj7410, 2019-11-25:
Which demo are you trying to run and from where are you following the instructions ?

Trancez15, 2019-11-25:
I’m following this website: https://pimylifeup.com/raspberry-pi-google-assistant/

manoj7410, 2020-05-12:
If you are still facing any issue then it is recommended to follow the instructions from https://aiyprojects.withgoogle.com/voice/ which is the official website of AIY projects.

closed 2020-05-13 by dmitriykovalev