google/aiyprojects-raspbian #653: why the error???

jamessong, 2019-11-24:
./assistant_library_demo.py 
Traceback (most recent call last):
  File "./assistant_library_demo.py", line 71, in <module>
    main()
  File "./assistant_library_demo.py", line 65, in main
    with Board() as board, Assistant(credentials) as assistant:
  File "/home/pi/AIY-projects-python/src/aiy/assistant/library.py", line 76, in __init__
    self._model_id = device_helpers.register_model_id(credentials)
  File "/home/pi/AIY-projects-python/src/aiy/assistant/device_helpers.py", line 93, in register_model_id
    r.raise_for_status()
  File "/usr/lib/python3/dist-packages/requests/models.py", line 940, in raise_for_status
    raise HTTPError(http_error_msg, response=self)
requests.exceptions.HTTPError: 400 Client Error: Bad Request for url: https://embeddedassistant.googleapis.com/v1alpha2/projects/aiy-2019-26X020/deviceModels/aiy-2019-26X020-voice-kit

manoj7410, 2019-11-25:
Did you try to run assistant_grpc_demo.py ?
Have you allowed your Google account to be used for assistant services ?

jamessong, 2019-11-25:
Did you try to run assistant_grpc_demo.py ?    yes, that works fine.

Have you allowed your Google account to be used for assistant services ?   - not sure how?