google/aiyprojects-raspbian #654: 2019-11-13 image on buster error

rickmini, 2019-11-29:
I am using the latest image from 2019-11-13 Raspberry Buster on both a Raspi 4 and 3b+. No matter what i do i cannot get past authentication.
1. i tried using my previous working assistant.json file installed in /home/pi which worked with Stretch on a 3b+. I also tried creating a new project with new oath creds and new auth screen. On any of these i get the following when i run a demo: for example  ./assistant_library_demo.py
or python3 assistant_library_demo.py

Traceback (most recent call last):
  File "./assistant_library_demo.py", line 71, in <module>
    main()
  File "./assistant_library_demo.py", line 64, in main
    credentials = auth_helpers.get_assistant_credentials()
  File "/home/pi/AIY-projects-python/src/aiy/assistant/auth_helpers.py", line 151, in get_assistant_credentials
    return _try_to_get_credentials(credentials_file)
  File "/home/pi/AIY-projects-python/src/aiy/assistant/auth_helpers.py", line 123, in _try_to_get_credentials
    credentials = _credentials_flow_interactive(client_secrets)
  File "/home/pi/AIY-projects-python/src/aiy/assistant/auth_helpers.py", line 75, in _credentials_flow_interactive
    webbrowser.register('chromium-browser', None, webbrowser.Chrome('chromium-browser'), -1)

rickmini, 2019-11-29:
i forgot to add these lines:
  File "/home/pi/AIY-projects-python/src/aiy/assistant/auth_helpers.py", line 75, in _credentials_flow_interactive
    webbrowser.register('chromium-browser', None, webbrowser.Chrome('chromium-browser'), -1)
TypeError: register() takes from 2 to 3 positional arguments but 4 were given

alessandrobaldassari, 2019-11-30:
You have to edit the auth_helpers.py file at line 75 as below:
`webbrowser.register('chromium-browser', None, webbrowser.Chrome('chromium-browser'))`

rickmini, 2019-11-30:
i did that, ran ./assistant_library_demo.py  , then got the auth screen,
and it completed authorization:  Then it exited and and left output:
Segmentation fault.

here is the outpu:

./assistant_library_demo.py

Please visit this URL to authorize this application:
https://accounts.google.com/o/oauth2/auth?response_type=code&client_id=23695939617-49irrcmkf3d8vms7i1n2si1u4g8ne3vs.apps.googleusercontent.com&redirect_uri=http%3A%2F%2Flocalhost%3A8080%2F&scope=https%3A%2F%2Fwww.googleapis.com%2Fauth%2Fassistant-sdk-prototype&state=LfjzwuQuw7vvFRlsv9RU8Dfve3480k&access_type=offline
INFO:google_auth_oauthlib.flow:"GET
/?state=LfjzwuQuw7vvFRlsv9RU8Dfve3480k&code=4/twGkL37URqEY3XwdNzYVuKSqgMS0XRgMAi6dGa1G_XtIdm0JQ6TUxk0sAJQ1VpPs7RCtj4NzuqiVrmLDuDGOwG4&scope=
https://www.googleapis.com/auth/assistant-sdk-prototype HTTP/1.1" 200 65
INFO:root:OAuth credentials initialized:
/home/pi/.cache/voice-recognizer/assistant_credentials.json
Segmentation fault

rickmini, 2019-12-04:
Any idea on why segmentation fault?
INFO:root:OAuth credentials initialized:
/home/pi/.cache/voice-recognizer/assistant_credentials.json
Segmentation fault

alessandrobaldassari, 2019-12-04:
I have the sane issue

Cordiali saluti,

Alessandro Baldassari
+39 338 1937531
alessandro.baldassari@live.it

rickmini, 2019-12-04:
Alessandro
I had assumed you were a google tech helping me. But i guess that you and I are just suffering the same frustration.
Does google ever respond to issues with their code?

manoj7410, 2019-12-13:
@rickmini Assistant library SDK has been deprecated and assistant_library_demo is broken at backend now. assistant_grpc_demo is running fine though.

mdbrewer, 2020-03-13:
Check out my writeup on how to get it working and update stretch. https://medium.com/@michaeldbrewer/aiy-voice-deprecation-workaround-bd07559403df

daruvurisai, 2020-03-14:
pip3 install google-assistant-library==1.0.0

use that to update the Library

dmitriykovalev, 2020-05-13:
This seems to be a duplicate of #660, please comment there.

closed 2020-05-13 by dmitriykovalev